don/cordova-plugin-ble-central #913: ble.startNotification on IOS , notifications are sent even if value doesn't change

playrobotics-alex, 2022-04-19:
Hi!
I am using ble.startNotification in order to subscribe to notifications.
On Andorid works great(I am only getting notifications when there is a change in value), but on IOS I am getting notifications with the exact same data over and over again.
Am I doing anything wrong?

 this.ble.startNotification(this.ID,SERVICE_UUID, "7E400003-B5A3-F393-E0A9-E50E24DCCA9E").subscribe(
            (data) => {
              this.onNotify(data);              
            },
            (error) => console.error('Unexpected Error', 'Failed to subscribe')
          )

peitschie, 2022-04-20:
That looks ok to me @playrobotics-alex 

What should trigger the notifications? Is it in response to a write of some sort or related to physical interactions with the device?

If you haven't already checked, it might be worth checking if you get the same behaviour with [LightBlue](https://apps.apple.com/au/app/lightblue/id557428110) or a similar app, just in case it's something to do with the peripheral behaviour.

playrobotics-alex, 2022-04-20:
Thanks, the peripheral  is ESP32 and it responds to write from the phone.        
          'pCharacteristic->setValue(value);
          pCharacteristic->notify();'
once the above code is exectuted by the ESP32(only once) - the notifications don't stop coming (all with the same data)

peitschie, 2022-04-20:
Right!
It's actually a bit simpler than that... iOS doesn't provide a way to distinguish whether the characteristic has updated due to a notification or due to a read value. See https://github.com/don/cordova-plugin-ble-central/blob/master/src/ios/BLECentralPlugin.m#L706

Android on the other hand has different methods for notify vs. read: https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L415

So... in essence, if you're triggering a read on the same characteristic you have hooked up a notify for, it looks like on iOS this causes the loop you're seeing.

playrobotics-alex, 2022-04-20:
I found out that the object was empty because buffer[0] should be used instead of buffer like explained here:
https://github.com/don/cordova-plugin-ble-central/issues/805
I think it might be worth to update the docs at the ble.startNotification section :-)

peitschie, 2022-05-02:
Glad you were able to solve the issue @playrobotics-alex 

Yeah... the ionic wrapper (not maintained by me) has some unexpected quirks that I don't really understand 😢