don/cordova-plugin-ble-central #914: Decreasing latency

playrobotics-alex, 2022-04-26:
Hi!
I am using this library in order to drive an RC car over BLE.
Obviously for such usage minimizing latency is very important.
Currently, I am sending the following string from the phone to the car(ESP32): XX|YY - XX is speed and YY is steering angle (for example 40|78).
I know that behind the scenes my string might be divided to several packets and each packet has its own latency, but how can I figure out what is my latency and also also how can I improve it?
Thanks!!

peitschie, 2022-05-02:
There's very little that you need to worry about in the plugin itself... most of this is related to Bluetooth connection interval settings.

If you're running on Android, things like connection priority can help (https://github.com/don/cordova-plugin-ble-central#requestconnectionpriority), but the key bit here is likely to be your connection interval: https://www.novelbits.io/ble-connection-intervals/

These generally need cooperation between the peripheral and the phone to get a fast (and stable!) connection.

Also, if you aim to keep your strings below the maximum transmission unit (usually this is done by keeping the payload size less than 100 bytes to be safest), you can keep each command to a single packet, which will make a big difference in the long run.

peitschie, 2022-07-07:
I'm closing this out as there's no obvious improvements needed to the plugin.
However, feel free to continue the discussion here if necessary.

closed 2022-07-07 by peitschie