don/cordova-plugin-ble-central #916: unwanted disconnect when iOS app goes to background

kevingre, 2022-05-25:
Having an issue only on iOS where when the app is put into the background, we are disconnected from the peripheral immediately.  However, scanning still works in the background and we auto-reconnect, only to be disconnected again.  This happens over and over until the app is foregrounded again.  I have the info.plist set:
`<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>`
as well as the background mode of "Uses Bluetooth LE accessories".  Also have properly set the bluetooth_restore_state:
`<preference name="BLUETOOTH_RESTORE_STATE" value="true" />`.

Here's some log entries from xcode - any ideas would be MOST appreciated!

This log snippet starts right after I put the app into the background:

2022-05-24 23:15:24.485028-0500 App[72015:6255900] didDisconnectPeripheral
2022-05-24 23:15:24.485365-0500 App[72015:6255900] Cleared notification callback BLE575733136 for key A398F816-346A-53DB-09E5-7172A4F820A3|C8100001-85E3-4E01-A7A2-85017037CA47|C8100008-85E3-4E01-A7A2-85017037CA47
To Native Cordova ->  BLE stopNotification BLE575733151 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3, C8100001-85E3-4E01-A7A2-85017037CA47, C8100008-85E3-4E01-A7A2-85017037CA47]]
2022-05-24 23:15:24.502925-0500 App[72015:6255900] stop notification
2022-05-24 23:15:24.503255-0500 App[72015:6255900] getData
2022-05-24 23:15:24.503556-0500 App[72015:6255900] Could not find service with UUID C8100001-85E3-4E01-A7A2-85017037CA47 on peripheral with UUID A398F816-346A-53DB-09E5-7172A4F820A3
⚡️  [log] - Disconnected from A398F816-346A-53DB-09E5-7172A4F820A3
⚡️  [log] - Dozer 2 disconnected.
To Native Cordova ->  BLE stopScan BLE575733152 ["options": []]
2022-05-24 23:15:24.537521-0500 App[72015:6255900] stopScan
To Native Cordova ->  BLE stopNotification ⚡️  [log] - Stopping notifications on DOZERSYNC_TELEMETRY_POSITION_CHARACTERISTIC_UUID
BLE575733153 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3, C8100001-85E3-4E01-A7A2-85017037CA47, C8100008-85E3-4E01-A7A2-85017037CA47]]
2022-05-24 23:15:24.538459-0500 App[72015:6255900] stop notification
2022-05-24 23:15:24.538606-0500 App[72015:6255900] getData
2022-05-24 23:15:24.538829-0500 App[72015:6255900] Could not find service with UUID C8100001-85E3-4E01-A7A2-85017037CA47 on peripheral with UUID A398F816-346A-53DB-09E5-7172A4F820A3
To Native Cordova ->  BLE disconnect BLE575733154 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3]]
2022-05-24 23:15:24.539170-0500 App[72015:6255900] disconnect
⚡️  [error] - ["Peripheral disconnected"]

⚡️  ------ STARTUP JS ERROR ------

⚡️  Peripheral disconnected
⚡️  URL: capacitor://localhost/polyfills.js
⚡️  polyfills.js:320:40

⚡️  See above for help with debugging blank-screen issues
To Native Cordova ->  BLE startScanWithOptions BLE575733155 ["options": [<__NSArrayM 0x2831d6490>(
C8100001-85E3-4E01-A7A2-85017037CA47
)
, {
    reportDuplicates = 0;
}]]
2022-05-24 23:15:24.570542-0500 App[72015:6255900] startScanWithOptions
⚡️  [error] - Unhandled Promise rejection: Could not find service with UUID C8100001-85E3-4E01-A7A2-85017037CA47 on peripheral with UUID A398F816-346A-53DB-09E5-7172A4F820A3 ; Zone: <root> ; Task: setTimeout ; Value: Could not find service with UUID C8100001-85E3-4E01-A7A2-85017037CA47 on peripheral with UUID A398F816-346A-53DB-09E5-7172A4F820A3 undefined
2022-05-24 23:15:36.032317-0500 App[72015:6255900] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = DozerSyncBox2;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "//8C";
        };
        kCBAdvDataServiceUUIDs =         (
            "C8100001-85E3-4E01-A7A2-85017037CA47"
        );
    };
    id = "A398F816-346A-53DB-09E5-7172A4F820A3";
    name = raspberrypi;
    rssi = "-47";
    state = disconnected;
}
⚡️  [log] - BLE Scan Result:  {"id":"A398F816-346A-53DB-09E5-7172A4F820A3","state":"disconnected","rssi":-47,"name":"raspberrypi","advertising":{"kCBAdvDataLocalName":"DozerSyncBox2","kCBAdvDataManufacturerData":{},"kCBAdvDataServiceUUIDs":["C8100001-85E3-4E01-A7A2-85017037CA47"],"kCBAdvDataIsConnectable":1}}
⚡️  [log] - iOS BLE Raw Ad Data: {"kCBAdvDataLocalName":"DozerSyncBox2","kCBAdvDataManufacturerData":{},"kCBAdvDataServiceUUIDs":["C8100001-85E3-4E01-A7A2-85017037CA47"],"kCBAdvDataIsConnectable":1}
⚡️  [log] - Raw MFG Data: ffff02
⚡️  [log] - BleAdvertisingData Parsed: {"mfgId":65535,"mfgData":2,"dozerId":2,"localName":"DozerSyncBox2","serviceUuid":"C8100001-85E3-4E01-A7A2-85017037CA47","connectable":1}
⚡️  [log] - Attempting to connect to Dozer: 2 using device address:A398F816-346A-53DB-09E5-7172A4F820A3
To Native Cordova ->  BLE connect BLE575733156 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3]]
2022-05-24 23:15:36.051022-0500 App[72015:6255900] connect
[... 225 lines cut ...]
⚡️  [log] - iOS BLE Raw Ad Data: {"kCBAdvDataLocalName":"DozerSyncBox2","kCBAdvDataManufacturerData":{},"kCBAdvDataServiceUUIDs":["C8100001-85E3-4E01-A7A2-85017037CA47"],"kCBAdvDataIsConnectable":1}
⚡️  [log] - Raw MFG Data: ffff02
⚡️  [log] - BleAdvertisingData Parsed: {"mfgId":65535,"mfgData":2,"dozerId":2,"localName":"DozerSyncBox2","serviceUuid":"C8100001-85E3-4E01-A7A2-85017037CA47","connectable":1}
⚡️  [log] - Attempting to connect to Dozer: 2 using device address:A398F816-346A-53DB-09E5-7172A4F820A3
To Native Cordova ->  BLE connect BLE575733177 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3]]
2022-05-24 23:15:56.847249-0500 App[72015:6255900] connect
2022-05-24 23:15:56.847409-0500 App[72015:6255900] Connecting to peripheral with UUID : A398F816-346A-53DB-09E5-7172A4F820A3
2022-05-24 23:15:58.292131-0500 App[72015:6255900] didConnectPeripheral
2022-05-24 23:15:58.736443-0500 App[72015:6255900] didDiscoverServices
2022-05-24 23:15:58.798913-0500 App[72015:6255900] didDiscoverCharacteristicsForService
2022-05-24 23:15:58.799520-0500 App[72015:6255900] Found characteristics for service <CBService: 0x282ac5900, isPrimary = YES, UUID = C8100001-85E3-4E01-A7A2-85017037CA47>
2022-05-24 23:15:58.799738-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98ae0, UUID = C8100002-85E3-4E01-A7A2-85017037CA47, properties = 0x1A, value = (null), notifying = NO>
2022-05-24 23:15:58.799924-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98a20, UUID = C8100003-85E3-4E01-A7A2-85017037CA47, properties = 0x2, value = (null), notifying = NO>
2022-05-24 23:15:58.800103-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98900, UUID = C8100006-85E3-4E01-A7A2-85017037CA47, properties = 0x2, value = (null), notifying = NO>
2022-05-24 23:15:58.800278-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98960, UUID = C8100007-85E3-4E01-A7A2-85017037CA47, properties = 0x2, value = (null), notifying = NO>
2022-05-24 23:15:58.800453-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b988a0, UUID = C8100008-85E3-4E01-A7A2-85017037CA47, properties = 0x12, value = (null), notifying = NO>
2022-05-24 23:15:58.800627-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98720, UUID = C8100004-85E3-4E01-A7A2-85017037CA47, properties = 0x8, value = (null), notifying = NO>
2022-05-24 23:15:58.800801-0500 App[72015:6255900] Characteristic <CBCharacteristic: 0x281b98780, UUID = C8100005-85E3-4E01-A7A2-85017037CA47, properties = 0x12, value = (null), notifying = NO>
⚡️  [log] - Connected to A398F816-346A-53DB-09E5-7172A4F820A3
⚡️  [log] - Dozer 2 connected.
⚡️  [log] - Starting notifications on DOZERSYNC_TELEMETRY_POSITION_CHARACTERISTIC_UUID
To Native Cordova ->  BLE startNotification BLE575733178 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3, C8100001-85E3-4E01-A7A2-85017037CA47, C8100008-85E3-4E01-A7A2-85017037CA47]]
2022-05-24 23:15:58.831475-0500 App[72015:6255900] registering for notification
2022-05-24 23:15:58.831571-0500 App[72015:6255900] getData
2022-05-24 23:15:58.831672-0500 App[72015:6255900] Looking for C8100008-85E3-4E01-A7A2-85017037CA47 with properties 16

peitschie, 2022-05-25:
@kevingre sounds a bit tricky! Is there any encryption on this connection? I have seen a symptom like this if iOS doesn't like the pairing keys being exchanged... it just eventually times out and disconnects the peripheral.

The other thing I'd check is whether this is being forced by the peripheral you're connecting to if it doesn't get the response it wants quickly enough?

peitschie, 2022-05-25:
@kevingre reading that log a bit more closely, there's a recurring `disconnect` message being logged prior to disconnection there, which I think corresponds to the `ble.disconnect` API being called by the app:
```
2022-05-24 23:15:53.032747-0500 App[72015:6255900] disconnect
```

The log line is emitted here:
https://github.com/don/cordova-plugin-ble-central/blob/1edda8f898a45bade498aeffa3862180bb0fd452/src/ios/BLECentralPlugin.m#L184

Do you call `disconnect` anywhere in your app? E.g., perhaps a window unload event or some kind of app shutdown step?

kevingre, 2022-05-25:
Tracing further, it looks like `didDisconnectPeripheral` is called in BLECentralPlugin.m (line 628) prior to the `disconnect` being called.  Since this isn't happening under on Android, I don't think there is code I specifically call that's making it happen.  I'm thinking it's a nuance with the iOS implementation.  Certainly nothing wrong with the plugin, more likely something I've misconfigured myself.  Or it may be something in Capacitor that's causing it only on the iOS side.  But since I am indeed able to scan and connect in the background, I feel like all the background mode stuff is configured correctly, just somewhere something is wanting to disconnect since I'm backgrounded.  I'm going to look further up the chain as it sounds like I've got the plugin configured correctly to work - just gotta figure out who/what is asking for a disconnect :)

kevingre, 2022-05-25:
No progress, but I did slim down the logs and turned off the reconnecting functionality so I could better discern what's happening.  Here's a slimmer log:

2022-05-25 16:29:15.478653-0500 App[77586:6691951] didDisconnectPeripheral
2022-05-25 16:29:15.478894-0500 App[77586:6691951] Cleared notification callback BLE729558111 for key A398F816-346A-53DB-09E5-7172A4F820A3|C8100001-85E3-4E01-A7A2-85017037CA47|C8100008-85E3-4E01-A7A2-85017037CA47
To Native Cordova ->  BLE stopNotification BLE729558112 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3, C8100001-85E3-4E01-A7A2-85017037CA47, C8100008-85E3-4E01-A7A2-85017037CA47]]
2022-05-25 16:29:15.496548-0500 App[77586:6691951] stop notification
2022-05-25 16:29:15.497071-0500 App[77586:6691951] getData
2022-05-25 16:29:15.497482-0500 App[77586:6691951] Could not find service with UUID C8100001-85E3-4E01-A7A2-85017037CA47 on peripheral with UUID A398F816-346A-53DB-09E5-7172A4F820A3
⚡️  [log] - Disconnected from A398F816-346A-53DB-09E5-7172A4F820A3
⚡️  [log] - Dozer 2 disconnected.
To Native Cordova ->  BLE disconnect BLE729558113 ["options": [A398F816-346A-53DB-09E5-7172A4F820A3]]
2022-05-25 16:29:15.532346-0500 App[77586:6691951] disconnect

⚡️  ------ STARTUP JS ERROR ------

⚡️  Peripheral disconnected
⚡️  URL: capacitor://localhost/polyfills.js
⚡️  [error] - ["Peripheral disconnected"]
⚡️  polyfills.js:320:40

⚡️  See above for help with debugging blank-screen issues

  The `didDisconnectPeripheral` inside BLECentralPlugin.m gets called and has an error of type "Unknown Error" which indicates the connection is not being terminated by a `disconnect` call.  Since this only happens in iOS, I'm thinking it's something to do with the background mode but I'm assuming others have gotten connections to stay active and receive notifications when in the background.

When I reconnect (when in the background) I can stay connected for 2.5 seconds on average.  And I can indeed receive notifications while connected.  I just can't STAY connected.

peitschie, 2022-05-26:
Very interesting!

In background mode, the application itself is only woken up for a few seconds after a Bluetooth event (connect/disconnect/discovery/etc.). But... I've definitely been able to keep the connection alive despite the app being killed in the background.

Do you have any other Bluetooth devices handy to see if it's specific to the peripheral you're testing against?

kevingre, 2022-05-26:
@peitschie That's what I was thinking as well.  I only have the single device on hand as this is custom hardware that has not yet been released and I can't get another one for a few days, but I did go ahead and write a BLE solution from scratch using the iOS sdk and am able to reproduce the issue.  As such, it definitely looks like a peripheral problem.  I'm going to need to break out the bluetooth sniffer to figure this one out but suffice to say it is definitely not the plugin or the configuration thereof.  Thanks a bunch for helping me on this.  I'll close this issue but let you know when I finally figure out what's up :)

closed 2022-05-26 by kevingre

peitschie, 2022-05-27:
Right. Both glad and non-glad to hear you've been able to reproduce with a standalone app.

Best of luck.. definitely interested in the outcome regardless 😸