don/cordova-plugin-ble-central #917: startScan() without filters doesn't allow me to scan in background

paesedombre, 2022-06-09:
There is any option to set an empty list of service UUID or filter for manufacturer data?  I want decode a iBeacon frame when i turn off the screen but Android doesn't allow.

peitschie, 2022-06-09:
Hello! For iBeacon scanning specifically, this plugin probably won't be the best fit as it can't scan by manufacturer data (iOS has no equivalent interface that I'm aware of).

You can attempt to scan with an empty service UUID list, but many Android manufacturers will block background scanning like this.

paesedombre, 2022-07-18:
It might help someone, I was able to make the app work in the background by adding a filter (for iBeacon) in the file node_modules > cordova-plugin-ble-central > src > android > BLECentralPlugin.java

and inserted these instructions at about line 980:

```
discoverCallback = callbackContext;
bluetoothLeScanner = bluetoothAdapter.getBluetoothLeScanner();
List<ScanFilter> filters = new ArrayList<ScanFilter>();
ScanFilter.Builder builder = new ScanFilter.Builder();
                builder.setManufacturerData(0x004c, new byte[] {});
                ScanFilter filter = builder.build();
                filters.add(filter);
bluetoothLeScanner.startScan(filters, scanSettings, leScanCallback);
```

peitschie, 2022-07-18:
Handy to know @paesedombre ... thanks for sharing!

If you feel so inspired, I'd be happy to have a patch that adds support for this to the `startScanWithOptions` method: https://github.com/don/cordova-plugin-ble-central#startscanwithoptions

There are a bunch of Android-specific ones there already, so adding an option like this would be fine:
```
"manufacturerData": {
  "id": 10,
  "data": ArrayBuffer,
  "dataMask": ArrayBuffer
}
```