don/cordova-plugin-ble-central #921: StartNotification doesn't return me  callback success or callback error

AngeNew, 2022-06-20:
Hi, I use startNotification to handle measure of IMU sensor, at startNotification sensor should me a constant signal of measures, I don't understand because startNotification tell me getData, but after doesn't return me callback success or  callback error. How can I trouble??
this is my code:

startMeasureNotification() {
    let buffer = new ArrayBuffer(3)
    let dataViewObject = new DataView(buffer)
    dataViewObject.setUint8(0, 0x01);
    dataViewObject.setUint8(1, 0x01);
    dataViewObject.setUint8(2, 0x21);
    this.ble.startNotification(this.id1, "15172000-4947-11e9-8646-d663bd873d93", "15172003-4947-11e9-8646-d663bd873d93").subscribe((result) => {
      console.log(result);
      this.ble.write(this.id1, "15172000-4947-11e9-8646-d663bd873d93", "15172001-4947-11e9-8646-d663bd873d93", dataViewObject.buffer).then((result) => {

        console.log(result);

      });

    }, (error) => {
      console.log(error);
    })
    
    this.ble.startNotification(this.id2, "15172000-4947-11e9-8646-d663bd873d93", "15172003-4947-11e9-8646-d663bd873d93").subscribe((result) => {
      console.log(result);
      this.ble.write(this.id2, "15172000-4947-11e9-8646-d663bd873d93", "15172001-4947-11e9-8646-d663bd873d93", dataViewObject.buffer).then((result) => {
        console.log(result);
      });
    }, (error) => {
      console.log(error);
    });

  }

peitschie, 2022-06-21:
The `getData` log there is a normal part of the `startNotification` flow:

https://github.com/don/cordova-plugin-ble-central/blob/0732380e27772a08df2c6ce6447b4cab901ad4d2/src/ios/BLECentralPlugin.m#L300

Does the Peripheral automatically start sending notifications as soon as you turn them on? Or do you need to write a payload to start them?

I ask because the `ble.write` you've nested in the `subscribe` there will only get called after a notification arrives.

AngeNew, 2022-06-21:
I need to write payload to start them, the ble.write  inside subscribe do start of measure, Do I can try this.ble.write before startNotification??

peitschie, 2022-06-24:
Apologies @AngeNew , this fell off my radar.

Yes, I'd do the write and then subscribe. What's the console output of `result` there?

I'd suggest taking a look at https://github.com/don/cordova-plugin-ble-central/issues/913#issuecomment-1103876933 in case this is a similar display issue.

Alternatively, you might be experiencing https://github.com/don/cordova-plugin-ble-central/issues/805#issuecomment-713639126

Unfortunately, I don't maintain the ionic wrappers, so I'm not certain what's causing the problem you see there.

AngeNew, 2022-06-24:
I trouble problem in this way, I use async await in right way, and I must convert my payload in hex to start correctly payload, thanks for your support

async startMeasure() {
    let buffer = new ArrayBuffer(3);
    let dataViewObject = new DataView(buffer);
    /** 01 indicate measures */
    dataViewObject.setUint8(0, 0x01);
    /** 01 indicate start measures */
    dataViewObject.setUint8(1, 0x01);
    dataViewObject.setUint8(2, 0x15);

    this.ble.startNotification(this.id1, "15172000-4947-11e9-8646-d663bd873d93", "15172003-4947-11e9-8646-d663bd873d93").subscribe(
      (data) => {
        console.log(this.saveSingleMeasure(data));
        this.measures1.push(this.saveSingleMeasure(data));
      },
      () => console.log("error")
    );

    await this.ble.write(this.id1, "15172000-4947-11e9-8646-d663bd873d93", "15172001-4947-11e9-8646-d663bd873d93", dataViewObject.buffer).then((result) => {
      console.log(result);
    }, (error) => {
      console.log(error);

    });

    this.ble.startNotification(this.id2, "15172000-4947-11e9-8646-d663bd873d93", "15172003-4947-11e9-8646-d663bd873d93").subscribe(
      (data) => {
        console.log(this.saveSingleMeasure(data));
        this.measures2.push(this.saveSingleMeasure(data));
      },
      () => console.log("error")
    );

    await this.ble.write(this.id2, "15172000-4947-11e9-8646-d663bd873d93", "15172001-4947-11e9-8646-d663bd873d93", dataViewObject.buffer).then((result) => {
      console.log(result);
    }, (error) => {
      console.log(error);

    });
  }

I trouble also problem that send me empty notifications with this function where I use Float32 right formar for my data
  /** save single measure in Float32   */
  saveSingleMeasure(buffer) {
    var data = new Float32Array(buffer[0]);
    return data;
  }