don/cordova-plugin-ble-central #924: can't get device name

aseniousarsene, 2022-07-08:
the objects i m getting after scanning dont have BLE device name can you please assist. thanks

peitschie, 2022-07-08:
@aseniousarsene what Operating System and version are you on? Is this Android 12 by chance...?

aseniousarsene, 2022-07-08:
Hi @peitschie thanks for the prompt reply, i m using android 10 . As you can see below i'm only getting the id,mac and rssi

I/Capacitor/Console: File: http://localhost/7333.f2fd3a4aacacc404.js - Line 1 - Msg: scanned device :{"id":"49:27:A5:B9:B4:A8","mac":"49:27:A5:B9:B4:A8","rssi":-56}

peitschie, 2022-07-11:
@aseniousarsene are you able to check if you get the name using another Bluetooth scanner? The name itself is a field that the Bluetooth peripheral needs to put into the advertising packet in order for the phone to find it. If you aren't seeing it here, it likely means your peripheral doesn't have this flag specified.

I'd suggest using [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to confirm this is the case.

peitschie, 2022-08-07:
This is possibly fixed by requesting the correct permissions on Android 12.
If the issue still occurs, feel free to re-open this issue.

closed 2022-08-07 by peitschie