don/cordova-plugin-ble-central #925: Duplicate permissions in AndroidManifest.xml due to conflicts between plugins

peitschie, 2022-07-08:
Combining Android 12 support with https://github.com/dpa99c/cordova-diagnostic-plugin results in duplicate `android.permission.BLUETOOTH` entries being introduced, due to the new constraints this plugin puts on `android:maxSdkVersion`.

Unfortunately, there's no automated solution to this as it's entirely unclear which arrangement should take priority. This conflict can't be addressed via `config.xml` only changes either, as `edit-config` only works on the first child element that matches, so there's no way to target the duplicate permission.

The best opt-in solution I've found is a `after_prepare` hook, that can be added to `config.xml` which post-processes the manifest and removes the duplicates.

I've included one in this plugin for easy reference, which can be activated by adding the appropriate hook to `config.xml`:
```
<?xml version='1.0' encoding='utf-8'?>
<widget>
    ...
    <platform name="android">
        <hook src="plugins/cordova-plugin-ble-central/stripDuplicatePermissions.js" type="after_prepare" />
    </platform>
</widget>
```

peitschie, 2022-07-11:
Workaround has been provided in #925.
No further fix for this can be done.

closed 2022-07-11 by peitschie

peitschie, 2022-10-04:
Hi @infacto 

It's not entirely clear to me what the issue you've encountered here is?

You are mostly correct in that usually Android happily merges duplicate permissions together. However, this behaviour does not apply if plugins t set different attributes on the permission.

With Android 12 introducing new flags regarding location, and obsoleting old permissions, I very quickly found that plugins which weren't actively maintained would cause compilation issues when used along side this up-to-date plugin.

peitschie, 2022-11-08:
To allow proper permissions handling, there is now a `slim` variant of this plugin which removes **all** Android permissions from the plugin itself.

This variant can be installed like this:
```
cordova plugin add cordova-plugin-ble-central@slim
```

Once this has been installed, the desired permissions need to be added back into the app's `AndroidManifest.xml` file via a `config-file` entry in the `config.xml`.

The default set of permissions the plugin installs on Android SDK v31+ can be added like this:

```
<config-file target="AndroidManifest.xml" parent="/manifest">
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
</config-file>
```

peitschie, 2022-12-02:
@andreacab2 sorry, I'm not quite following the problem there? 😅

If you use the normal plugin, this is pretty much expected to happen. The `after_prepare` hook noted in https://github.com/don/cordova-plugin-ble-central/issues/925#issue-1298493141 may be sufficient to fix the problem, but you need to add it to `config.xml` yourself in order to activate this feature.

If you are using the `slim` variant and manually specifying the Bluetooth permissions yourself, you'll want to remove the `<uses-permission android:name="android.permission.BLUETOOTH" ... />` fragment from your own specified `config.xml` as well.

BenLaKnet, 2023-06-30:
Thanks for:

<platform name="android">
        <hook src="plugins/cordova-plugin-ble-central/stripDuplicatePermissions.js" type="after_prepare" />
    </platform>

it is running for me.