don/cordova-plugin-ble-central #926: Version 1.5.0 has breaking changes

nunoarruda, 2022-07-18:
Support for Android 12 (API level 31) is a breaking change since projects not yet using target SDK 31 will fail to build:

```
Android resource linking failed
/Users/nunoarruda/repos/Pulse-Angular/android/app/build/intermediates/packaged_manifests/debug/AndroidManifest.xml:31: error: attribute android:usesPermissionFlags not found.
error: failed processing manifest.
```

If you folks are following [semver](https://semver.org/) then the 1.5.0 release should have been a major version bump like 2.0.0.

Sidenote: I've tried updating the target SDK from 30 to 31 and, while that fixed the issue, using target SDK 31 introduced other errors, at least for me or for my project. Sticking to version 1.4.4 for now.

peitschie, 2022-07-18:
Hi @nunoarruda 

What build tooling are you using there? That path you are showing there isn't a vanilla cordova platform at all.

There are no breaking changes in this release assuming the build tool you're using correctly respects the `after_prepare` cordova hook. Unfortunately, Android 12 demands a bunch of XML attributes that are incompatible with previous Android SDKs, so an `after_prepare` script is employed to strip out these unsupported permissions and attributes.

See https://github.com/don/cordova-plugin-ble-central/blob/master/hooks/after_prepare.js for the details of this hook.

If you let me know what tool you're using, perhaps we can find a solution that is equivalent there as well.

nunoarruda, 2022-07-18:
@peitschie thanks for the feedback

I'm using [Capacitor](https://capacitorjs.com/) so perhaps it only breaks in Capacitor projects.

If you're interested in supporting Capacitor then a good place to start might be this page: https://capacitorjs.com/docs/cordova#plugin-management

If you're focusing on supporting only Cordova I won't blame you and you can close this issue :)

peitschie, 2022-07-18:
Thanks for the link... I'm definitely open to fixes that can improve the integration with Capacitor.

At a glance, it's tricky to figure out how the permissions from the library get merged into the main Android manifest. Android Bluetooth permissions between SDK 30 and SDK 31 are completely incompatible, so this plugin has had to do some clever things in order to support Cordova apps targeting both. Of specific interest is that I need to use both the target SDK version of the app AND the OS platform version in order to determine which permissions to request: https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L1053

Though the hooks don't function with capacitor, in theory, by leveraging libraries Capacitor opens up the possibility of your app using the standard Android manifest merge operations to control removing the SDK 31 permissions and attributes that you don't want. You might be able to use the `tools:remove` flag to eliminate the problematic attribute or `tools:node="remove"` to eliminate the Android 12 permissions completely.

See https://developer.android.com/studio/build/manage-manifests#merge-manifests for tips about these (if you weren't already aware).

Also worth noting that by the end of this year all new apps and even application updates to the Play Store will [require SDK 31 at a minimum](https://developer.android.com/google/play/requirements/target-sdk).

But... keep me posted with what you find, as I'd be interested in what solution ends up working best for your situation.

nunoarruda, 2022-08-03:
Tip: Capacitor just released [v4](https://ionicframework.com/blog/announcing-capacitor-4-0/) which now sets a deployment target of Android 12 (SDK 32). `cordova-plugin-ble-central` v1.5.0 is working fine with Capacitor v4.

peitschie, 2022-08-05:
Handy to know! Thanks for the update @nunoarruda

peitschie, 2022-08-09:
I'll close this one out as upgrading capacitor seems like a reasonable fix for this.

closed 2022-08-09 by peitschie