don/cordova-plugin-ble-central #928: `ACCESS_BACKGROUND_LOCATION` permission is being added by default

nunoarruda, 2022-08-08:
Tested on v1.5.0+ with [Capacitor](https://capacitorjs.com/) v4+. This was previously working fine with v1.4.4 with Capacitor v3.

I was expecting this permission to be optional or turned off by default. Google is starting to complain about it, like in #870 

Does anyone have any idea what could be the issue here? Is it because of Capacitor v4? Is it because of any recent changes to this plugin?

Thanks in advance.

peitschie, 2022-08-08:
@nunoarruda that's an unintentional impact from the recent Android 12 changes I've made here. I was previously unaware that capacitor fails to run the after_install hooks, which is what's used to strip out the undesired `ACCESS_BACKGROUND_LOCATION` setting.

I'll have a look and see how hard it is to invert this logic so the cordova hook adds the permission in, rather than stripping it out.

peitschie, 2022-08-09:
@nunoarruda are you able to spin up a quick example repository showing this on a clean capacitor 4 project? Or give me a quick set of commands to achieve this issue?

I've attempted to spin up a Capacitor v4 project and added just this plugin, but it does not seem like that results in any changes to the `AndroidManifest.xml` directly, so I can only assume I'm missing something here.

peitschie, 2022-08-09:
Confirming my understanding here, Capacitor runs none of the hooks and copies none of the data from the `plugin.xml` file. I seems like the `ACCESS_BACKGROUND_LOCATION` should not be making it's way into your project via any automatic fashion, so I'm puzzled about the issue you're seeing here?

Reference: https://capacitorjs.com/docs/plugins/cordova#variables-and-hooks

Is the permission appearing in your `AndroidManifest.xml`? Or is the complaint from Google about the runtime permission appearing on code?

The `ACCESS_BACKGROUND_LOCATION` string does appear in code, but is only used if you specifically opt in by setting the `ACCESS_BACKGROUND_LOCATION` config flag to `true`.

peitschie, 2022-08-09:
Right... managed to get all the way through this.
Basically, I was on the right path, just needed to do the `gradle` call to build the app after installing the plugin.

There's a workaround for what you're seeing @nunoarruda in that you can use Manifest Merging to remove this unwanted element, by adding this line to your main `AndroidManifest.xml` file:
```
<manifest xmlns:tools="http://schemas.android.com/tools" ...>
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" tools:node="remove"/>
</manifest>
```

But... I strongly suspect very few people are using this feature, so I'll make some changes for a future release to try and remove this sharp edge.

peitschie, 2022-08-09:
I've released a fix for this @nunoarruda 

If the problem still occurs with 1.5.2, feel free to re-open this issue.

closed 2022-08-09 by peitschie

nunoarruda, 2022-08-09:
The `ACCESS_BACKGROUND_LOCATION` permission is no longer included by default when using v1.5.2.

Thanks for digging into this.

PR #929 (2022-08-09): Exclude ACCESS_BACKGROUND_LOCATION from plugin.xml to avoid complications with Capacitor #928
Fixed #928