don/cordova-plugin-ble-central #930: iOS appears to stop receiving advertisements after a while

huiigary, 2022-08-23:
Hi,

Was wondering if anyone noticed an issue on their iPhone where the using BLE.startScanWithOptions() appears to stop receiving advertisements? Also tested on Android, but I don't seem to observe this issue there.

Also, is it possible for the BLE scan to receive all the advertisements from a device? I seem to be missing advertisements and Im not sure if it is a Cordova issue or a general issue with  phones unable to receive all advertisements. 
Even using nRF-connect, it seems to be missing advertisements too which suggests that phones are unable to receive all advertisements.

    "@ionic-native/ble": "^5.31.1",
    "cordova-plugin-ble-central": "^1.4.1",

favnec5, 2022-08-24:
Hi,
Same thing here on last version of IOS. After ble connexion, something crash. the beacon became "hide" into the scan. it work in an other app like nRF Connect and with cordova-plugin-bluetoothle.
I don't understant the very idea of "blacklist" a beacon. Could this be related to Address iOS deprecations and issues #919 ?
This is a critical issue for IOS :-/
Best regards
Thomas

huiigary, 2022-08-24:
@favnec5 You tried using Cordova-plugin-bluetoothLE and you don't observe the issue of iOS appearing to stop receiving advertisements? 

Yea it is strange because on nRF-Connect is receiving advertisements at an acceptable rate (misses some adverts at the given advertising interval), however it doesn't appear to hang and not receive advertisements like what I'm observing.

I also have the iOS deprecation warnings sen in #919. I assumed because they are warnings and not errors, there would no significant effect on the plugin.

peitschie, 2022-08-25:
Are these scans being run via `startScanWithOptions` and setting the `reportDuplicates` flag to true? If just `scan` or `startScan` is being used, likely what's happening is that iOS will filter out packets it thinks are duplicates.

See doco for this here: https://github.com/don/cordova-plugin-ble-central#startscanwithoptions

favnec5, 2022-08-25:
Hi,
I'm still researching this problem.
Apparently, on my side, the problem is a difference between the localName and Name on the beacons concerned.
@huiigary,  sorry, so it may have nothing to do with your problem.
I fixed with this code :
`if(device.advertising && device.advertising.kCBAdvDataLocalName) device.name = device.advertising.kCBAdvDataLocalName ;`

huiigary, 2022-08-25:
@peitschie 
Yes on iOS I am using `startScanWithOptions`, but with `reportDuplicate=false` because I notice the scans returns duplicates advertisements received around same time. I will test using reportDuplicates=true to see if it fixes this issue.

However on Android, `startScanWithOptions` with `reportDuplicate=true`, to get advertisements from the peripheral. Otherwise I get a single advertisement but not other advertisements.

From testing my App, it appears to be an iOS specific issue where sometimes after ~2mins of scanning (times vary), it doesn't receive more advertisements for a while. Then maybe a few minutes later it resumes. I wonder if iOS has some internal timer that will stop scanning after some time.

I also did a quick test using another plugin https://github.com/randdusing/cordova-plugin-bluetoothle for scanning to see if it was an issue with this plugin: Cordova-BLE-central. Running their scanning, my iOS App didn't advertisements after sometime too. 

So far it appears to be an iOS issue or something to do with my Ionic/Angular App.

favnec5, 2022-08-25:
@huiigary 
Does your app stay "awake" or does it switch to screen off when scan is failing ?
Because the "background mode" is not really possible. 
In this eventuality, you can use https://github.com/EddyVerbruggen/Insomnia-PhoneGap-Plugin.

peitschie, 2022-08-26:
@favnec5 small correction, with iOS this plugin does allow background scanning to occur 🙂 

See https://github.com/don/cordova-plugin-ble-central/#background-scanning-and-notifications-on-ios for implementation details!

favnec5, 2022-08-26:
Yes you are right. What I mean is, don't just assume it's that easy.
I have been using this plugin for about 4 years.
On the one hand you will have to justify background to Apple to publish any App on the store (which is long but doable), on the other hand, the phone will voluntarily limit the scans to save energy (slower detections). When the battery is too low, there is no background scanning when the power saver is launched as the app simply stops. 
I correct my statement :  background mode is not always really possible.

huiigary, 2022-08-31:
@favnec5 My app stays "awake" (the screen is on). What I notice with other App like nRF-connect is that is also stops receiving advertisements after some time too (refer to image below). Next clicked the home screen then went back into nRF-connect and it started to scan again... (at 1058 timestamp)
[image]

The question is what is iOS doing that it seems to stop scanning? If I dont touch the phone for a while, does it enter some state Im not aware of?

@peitschie 
To install the app, I do:
```
$ npm install cordova-plugin-ble-central 
$ npm install @awesome-cordova-plugins/ble 
$ ionic cap sync 
```
Im unable to run the `cordova` commands as it results in an error. My guess because Im running a capacitor app. Therefore Im unable to run the command `$ cordova plugin add cordova-plugin-ble-central --variable BLUETOOTH_RESTORE_STATE=true
`
 to enable background scanning for iOS to see if it fixes my issue

favnec5, 2022-08-31:
I checked my app on my phone with ios 15.6.1 (cordova 11.0.0), the scan don't stop.
```
        ble.startScanWithOptions(["2A6E","2A6F","2AA1","2A06"], { "reportDuplicates": true }, findedDevice2, scanFailed);
```

Note : A long time ago, i added in config.xml file :
```
  <edit-config file="*-Info.plist" target="UIBackgroundModes" mode="replace">
    <array>
      <string>bluetooth-central</string>
      <string>location</string>
    </array>
  </edit-config>
```

linkitapps, 2023-08-12:
> I checked my app on my phone with ios 15.6.1 (cordova 11.0.0), the scan don't stop.
>
> ```
>         ble.startScanWithOptions(["2A6E","2A6F","2AA1","2A06"], { "reportDuplicates": true }, findedDevice2, [...]

hello
Does your app still work well on iOS?

I did the same, but now I'm asking because scanning in the background doesn't work on my iPhone.