don/cordova-plugin-ble-central #932: Incompatibility with the Geolocation or Location Accurancy plugins

Nickholas, 2022-09-07:
Probably, you will think that i am a little crazy but i have encountered an incompatibility between the BLE Central plugin and the Geolocation or Location Accurancy plugins of ionic. Its about permissions.

Its very strange but when i have the Geolocation and Location Acurrancy plugins installed all works fine BUT, when i install the BLE plugin in my solution, the app tolds me that i have not specified the required android permissions on my AndroidManifest.xml, despite i did it.

My current package.json is this:

`"dependencies": {
    "@angular/common": "^14.0.0",
    "@angular/core": "^14.0.0",
    "@angular/forms": "^14.0.0",
    "@angular/platform-browser": "^14.0.0",
    "@angular/platform-browser-dynamic": "^14.0.0",
    "@angular/router": "^14.0.0",
    "@awesome-cordova-plugins/android-permissions": "^5.45.0",
    "@awesome-cordova-plugins/app-version": "^5.44.0",
    "@awesome-cordova-plugins/barcode-scanner": "^5.45.0",
    "@awesome-cordova-plugins/bluetooth-serial": "^5.44.0",
    "@awesome-cordova-plugins/file": "^5.44.0",
    "@awesome-cordova-plugins/file-opener": "^5.44.0",
    "@awesome-cordova-plugins/location-accuracy": "^5.44.0",
    "@awesome-cordova-plugins/native-audio": "^5.44.0",
    "@awesome-cordova-plugins/network": "^5.44.0",
    "@awesome-cordova-plugins/nfc": "^5.45.0",
    "@awesome-cordova-plugins/sqlite": "^5.45.0",
    "@awesome-cordova-plugins/sqlite-porter": "^5.45.0",
    "@awesome-cordova-plugins/status-bar": "^5.45.0",
    "@capacitor/android": "4.0.1",
    "@capacitor/app": "4.0.1",
    "@capacitor/camera": "^4.0.1",
    "@capacitor/core": "4.0.1",
    "@capacitor/geolocation": "^4.0.1",
    "@capacitor/haptics": "4.0.1",
    "@capacitor/keyboard": "4.0.1",
    "@capacitor/push-notifications": "^4.0.1",
    "@capacitor/splash-screen": "^4.0.1",
    "@capacitor/status-bar": "4.0.1",
    "@ionic-native/android-permissions": "^5.36.0",
    "@ionic-native/app-version": "^5.36.0",
    "@ionic-native/location-accuracy": "^5.36.0",
    "@ionic-native/network": "^5.36.0",
    "@ionic-native/status-bar": "^5.36.0",
    "@ionic/angular": "^6.1.9",
    "@ionic/pwa-elements": "^3.1.1",
    "@types/leaflet": "^1.7.11",
    "cordova-plugin-android-permissions": "^1.1.4",
    "cordova-plugin-app-version": "^0.1.14",
    "cordova-plugin-nativeaudio": "^3.0.9",
    "cordova-plugin-network-information": "^3.0.0",
    "cordova-plugin-request-location-accuracy": "^2.3.0",
    "cordova-plugin-statusbar": "^3.0.0",
    "cordova-sqlite-storage": "^6.0.0",
    "jwt-decode": "^3.1.2",
    "leaflet": "^1.8.0",
    "leaflet-easybutton": "^2.4.0",
    "leaflet-iconmaterial": "^1.1.0",
    "leaflet-lasso": "^2.2.12",
    "moment": "^2.29.4",
    "ng2-search-filter": "^0.5.1",
    "ngx-echarts": "^14.0.0",
    "ngx-ionic-image-viewer": "^0.7.5",
    "phonegap-nfc": "^1.2.0",
    "phonegap-plugin-barcodescanner": "^8.1.0",
    "rxjs": "~6.6.0",
    "tslib": "^2.2.0",
    "uk.co.workingedge.cordova.plugin.sqliteporter": "^1.1.1",
    "underscore": "^1.13.4",
    "zone.js": "~0.11.4"
  },`

At that moment, all works fine. After the BLE instalation:

`$ npm install cordova-plugin-ble-central 
$ npm install @awesome-cordova-plugins/ble 
$ ionic cap sync`

the Geolocation tolds me that no have permissions:

`18787-18787/? E/GrantPermissionsViewModel: None of [android.permission.ACCESS_FINE_LOCATION, android.permission.ACCESS_COARSE_LOCATION] in {android.permission-group.STORAGE=[android.permission.READ_EXTERNAL_STORAGE, android.permission.WRITE_EXTERNAL_STORAGE], android.permission-group.NEARBY_DEVICES=[android.permission.BLUETOOTH_SCAN, android.permission.BLUETOOTH_CONNECT], android.permission-group.PHONE=[android.permission.READ_PHONE_STATE], android.permission-group.CAMERA=[android.permission.CAMERA]}?`

peitschie, 2022-10-02:
@Nickholas apologies for the delayed reply, this one got lost in my inbox somehow.

What does the merged AndroidManifest.xml look like in the plugins library with all these settings present?

If you haven't already, it might be worth attempting this with v1.4.2 of this plugin rather than 1.5.0, just to see if it's related to the fairly recent Android 12 permission changes.

ShrinivasanS, 2022-11-08:
@peitschie  ,I too faced similar issue and it happens only for Android 12. When I looked into the merged AndroidManifest.xml file it has "maxSdkVersion" as 28 for ACCESS_COARSE_LOCATION permission. I override the AndroidManifest.xml of cordova-plugin-ble-central npm package to use "maxSdkVersion" as 32 then it is working fine.

 But this fix involve overriding package files. Is there any other way to other than this?

peitschie, 2022-11-08:
Android certainly is not making life easy for us 😓 

@ShrinivasanS I've just created a new `slim` variant of this plugin which removes **all** Android permissions from the plugin itself, as I can't think of any other way to manage this adequately.

You can install this variant like this:
```
cordova plugin add cordova-plugin-ble-central@slim
```

Once this has been installed, you'll need to add the desired permissions back into your `AndroidManifest.xml` file via a `config-file` entry in your `config.xml`.

The default set of permissions the plugin installs on Android SDK v31+ can be added like this:

```
<config-file target="AndroidManifest.xml" parent="/manifest">
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
</config-file>
```

Please let me know if you encounter any further issues with this.

ShrinivasanS, 2022-11-08:
@peitschie , Thanks for the quick response. I will try with the package.

peitschie, 2022-11-10:
I'll close this one out as the `slim` variant should allow this problem to be solved.
Unfortunately, due to the limitations of the cordova plugin framework, there's nothing much else I can do to make this easier to deal with!

closed 2022-11-10 by peitschie