don/cordova-plugin-ble-central #934: Is this possible to read datas from samsung gear S2 ?

BenLaKnet, 2022-10-01:
Hi,

I want to know if it is possible to use this plugin to connect to Samsung gear, to obtain feets on a day and other datas.

Many thanks.

peitschie, 2022-10-02:
hi @BenLaKnet 

It's hard to know for certain without having a device to test with.

As a basic check, I'd suggest using [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to see if you can connect directly to the Samsung Gear and explore any published Bluetooth services. If you can see any services listed on nrf Connect, you'll be able to use this plugin to extract data (once you figure out the service implementations you need).

BenLaKnet, 2022-10-03:
@peitschie Many thanks,

I have to find an app into gear to view datas and after syncing with nrf Connect to dectect services.

peitschie, 2022-10-03:
Yep, basically. Though, ideally you shouldn't need any app other than nrf Connect to connect to the device.

BenLaKnet, 2022-10-15:
I found to make visible gear S2:

From the watch screen press the Home button
to open the apps screen.
Home button
Tap Settings
.
Tap Connections
.
Tap Bluetooth
.
Tap Bluetooth
to turn on On indicator or off Off indicator.
Your device is visible to other devices for 2 minutes as long as the Bluetooth® settings screen is open and Bluetooth is turned on.

it is ok with nrf Connect.

And now the services....

BenLaKnet, 2022-10-16:
I found model name on:
service_uuid		: "1800",
	characteristic_uuid : "2A00",
	affichetype			: "Modèle"

and now datas from number of steps and calories

BenLaKnet, 2022-10-18:
There is no solution at this time but I do not say my last word ;):

```
Short answer is No, it is not possible now.

Longer explanation is that in order to access the Health API you need to be a Samsung Health Partner (Tizen OS) and that is no longer open for applying. Samsung Partners are for Wear OS powered by Samsung developers not Tizen OS developers.

Ron
Samsung Developer Relations
```

peitschie, 2022-10-18:
Right... they don't seem terribly interested in giving you details!

BenLaKnet, 2022-10-18:
Hi,

I found smart watch and services with nrfConnect, but none with my code.
What's the difference between scan, autoscan, startScan ?

peitschie, 2022-10-18:
You won't see any services until you connect to the device after scanning it.

BenLaKnet, 2022-10-19:
I am going to try with a little smart watch : msgm115 / ID115.

peitschie, 2023-01-06:
Hi @BenLaKnet 

The difference (I'm guessing, as you haven't clarified) is that you have not just scanned with nrf Connect, but actually connected to the device.

If you scan and then connect to the device with this plugin, the behaviour should be the same.

I'll close this out for now as it's not clear there is a plugin issue here, but feel free to continue the discussion or reopen.

closed 2023-01-06 by peitschie