don/cordova-plugin-ble-central #935: scan some time not find any device

hdpklm, 2022-10-09:
Hello 
I use this plugin in cordova for android@10

my problem is some time scan cant find any device and other time work without any problem
and test in diferent mobiles and all mobiles has the same problem, and the scan work some time in some mobiles and some time in all mobiles.

I use this code 
`startScan([], console.log, console.warn)`

I find a lot of issues for scan but not sure if any one solution this problem
please help my

thank you.

peitschie, 2022-10-09:
Hi @hdpklm 

Are you able to post some code samples around this? What's the global object name you're calling `startScan` on?

Also, have you got checks in  your code to ensure location services are enabled, and the Bluetooth adapter itself is enabled?
Usually, before scanning, you'll need to run these two checks:
* https://github.com/don/cordova-plugin-ble-central#islocationenabled
* https://github.com/don/cordova-plugin-ble-central#isenabled

E.g.,:
```
if (!await ble.withPromises.isEnabled()) console.warn("Bluetooth adapter is not enabled");
if (!await ble.withPromises.isLocationEnabled()) console.warn("Location services are not enabled");
```

hdpklm, 2022-10-09:
`islocationenabled: Location services disabled.`
ok thank you. its work

peitschie, 2022-10-09:
Glad to hear it! I'll close this out for now, but feel free to re-open if you are still finding an issue.

closed 2022-10-09 by peitschie