don/cordova-plugin-ble-central #936: NullPointerException discoverServices after refreshDeviceCache

fardom, 2022-10-14:
Unfortunately there is a bug in `v1.5.2` which does not appear in `v1.3.1`. ([see diff](https://github.com/don/cordova-plugin-ble-central/compare/1.3.1...1.5.2))

In my case I updated an BLE device and its characteristics configuration. To avoid cache problems the method `refreshDeviceCache` is called after devices BLE reconfiguration.

```
Waiting 10000 milliseconds before discovering services
FATAL EXCEPTION: JavaBridge
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGatt.discoverServices()' on a null object reference
    at com.megster.cordova.ble.central.Peripheral$1.run(Peripheral.java:236)
    at android.os.Handler.handleCallback(Handler.java:938)
    at android.os.Handler.dispatchMessage(Handler.java:99)
    at android.os.Looper.loopOnce(Looper.java:226)
    at android.os.Looper.loop(Looper.java:313)
    at android.os.HandlerThread.run(HandlerThread.java:67)
```

https://github.com/don/cordova-plugin-ble-central/blob/a532e8b1efb438a63681e6ddfaf4ad0435de2d22/src/android/Peripheral.java#L236

fardom, 2022-10-14:
Thank you for the fast fix. I'll test it next monday. On first sight, I see that `discoverServices` will never called when GATT is `null`. I wonder why the previous version works here. And when is the GATT ready? I haven't dug into the source code yet. Maybe this is no problem, I'll scan after that anyway. Or we need to get the actual GATT again. Idk, I'll test it soon...

fardom, 2022-10-17:
Thanks. It looks good. No error here. But I wonder why this works in 1.3.1 and not in 1.5.2? The method is the same. I mean the GATT is null in callback after cache refresh. Not sure, but I think this can cause problem and require a disconnect, scan and connect again to indirectly trigger discoverServices. Otherwise we have characteristics with wrong reference. Because the BLE configuration of the peripheral has been changed and need to be discovered again. I need to test it with a disconnect, scan and connect again after refresh cache to validte it. Sigh, BLE testing is really exhausting...

peitschie, 2022-10-17:
> Thanks. It looks good. No error here. But I wonder why this works in 1.3.1 and not in 1.5.2?

Yeah, it is a good question.

Is this the first time you've targeted API 32 with this call/app? It's possible there's Android changes causing this. There's also been changes to this plugin to better clean-up gatts on disconnect, etc. to avoid stale gatts and dead object exceptions.

Are you disconnecting from the peripheral in the mean time? Does this exception happen on **every** refresh?

> Otherwise we have characteristics with wrong reference. Because the BLE configuration of the peripheral has been changed and need to be discovered again

Just in passing, there's a Service Changed characteristic that's part of the Bluetooth spec which can be used by as peripheral to indicate that services and attributes have been modified, potentially eliminating the need to use this refreshCache call. See https://stackoverflow.com/a/73524720/156169. I can appreciate you might not have control over this however 🙂 

> I'm also a bit confused about [this commit](https://github.com/don/cordova-plugin-ble-central/commit/4a481207e0b0a3a31e8a1109f5498fa675cebfeb). Connection callback loop

This fixes a crash if the app disconnects from the peripheral before service discovery has completed. What was happening is the app would call `ble.disconnect` during service discovery, which would set the `connectionCallback` to null. The service discovery would then complete, causing a null pointer exception.

> Sigh, BLE testing is really exhausting

Yep... I definitely can relate to your pain here! It's never an easy thing to confirm everything works correctly.

peitschie, 2022-10-17:
Just re-examining that commit a bit further, it does remove the override effect of the refresh device callback, meaning that the success path of `ble.connect` is conceivably called a second time with the new services.

Is there any chance this is causing your code to re-refresh the device cache again?

The removal of that override behaviour there is not intentional, and possibly should be reverted back.

peitschie, 2022-10-18:
@fardom I've pushed a new build up to cordova-plugin-ble-central@1.5.3-alpha.1 which reverts the refresh behaviour with regards to those callbacks back to 1.3.1.

Are you able to give this one more try through? If it's working for you there, I'll cut a new release with these changes in place.

fardom, 2022-10-18:
Thanks, it seems to work. I was able to update the peripheral without errors. ...

peitschie, 2022-10-18:
This should now be fixed in cordova-plugin-ble-central@1.6.0

commit 09e05cf7 (2022-10-18): Prevent NPE if device disconnects before device cache refresh #936

Fixes #936