don/cordova-plugin-ble-central #937: Android stopNotification causes errors if done too quickly

sam-higgs, 2022-10-14:
Upon calling `.stopNotification()` an error occurs when trying to do any other ble action too quickly afterwards e.g. `.startNotification()` or `.read()`

Has anyone else experienced this? Adding a delay of over 100ms seems to be a workaround.

peitschie, 2022-10-14:
Hi @sam-higgs 

Are you able to provide any logs with what the reported errors are?

Generally speaking, you should wait for the callbacks to complete (or await the `ble.withPromises.stopNotification(...)` promise) before starting another notification or reading another value.

Out of interest, what's the use-case here that's requiring you to stop and then start the notifications in quick succession?

graphefruit, 2024-08-11:
Hey @peitschie,
I am coming back to this issue.
Somehow I got a bluetooth scale which is doing strange things on customers side.
Most of the times, the app can connect to the bluetooth scale.
But sometimes, the app can't connect to the bluetooth scale.
The issue comes from the `ble.startNotification`
```
 "DEFAULT: [\"failed to subscribe to notifications \\\"Could not find service with UUID 49535343-fe7d-4ae5-8fa9-9fafd205e455 on peripheral with UUID 140FC66D-B212-05F1-F5C1-2A285331FC85\\\"\"]",
 ```
 
 If the user restarts the app, then the app can connect to the given UUID and Notification ID.
 Also to mention that when trying to `ble.connect` again after some seconds and try again to `ble.startNotification` don't resolve the issue, you realy need to restart the app.
 
 I've also tried to add a timeout with 150ms before calling `ble.startNotification` but this doesn't help either.
 
 Do you have any ideas why this behaves that strange?
 
 Thanks for your great work and your awesome support!
 Lars

peitschie, 2024-08-12:
Hi @graphefruit 

That error message is from iOS (https://github.com/don/cordova-plugin-ble-central/blob/7984913436429c850c1c5db25e86053d53530073/src/ios/BLECentralPlugin.m#L1153). Are you able to split this out into a separate issue for discussion?