don/cordova-plugin-ble-central #943: Legacy advertisement device not discovered when targeting Android 12

sam-higgs, 2022-11-09:
The following device is very rarely returned when targeting Android 12 (v1.5.2), previously was always found when using v1.4.1, any ideas? A basic bluetooth scanner app finds it all the time:

[image]

peitschie, 2022-11-10:
Interesting to know @sam-higgs ! Thanks for investigating this.

I've very recently created a `slim` variant of this package which removes the AndroidManifest.xml permissions completely, so an integrating application can specify whatever custom combinations are required. If you need to have the `neverForLocation` flag removed, I'd suggest this is the best way to achieve that goal.

To use this variant of the plugin instead, you can install the `cordova-plugin-ble-central@slim` package, and add a `config-file` element to `config.xml` with the exact Android permissions you want for your app. For example, the default set excluding the `neverForLocation` flag is:

```
<config-file target="AndroidManifest.xml" parent="/manifest">
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
</config-file>
```

peitschie, 2022-11-10:
I'll close this issue, as it seems like you've managed to figure out what causes the inconsistency. If you have any issues with the `slim` approach above (or have better ideas here!), feel free to continue the discussion here 🙂

closed 2022-11-10 by peitschie

sam-higgs, 2022-11-11:
@peitschie Yeah this is the approach we went for, seems to work - thanks for the quick response :)

hongwei98, 2023-03-14:
@peitschie i removed android:usesPermissionFlags, the scan not working anymore

peitschie, 2023-03-14:
@hongwei98 was it working prior to removing those flags?

hongwei98, 2023-03-15:
> 

yes, not removing the flags is working but it was not able to scan the legacy advertisement

peitschie, 2023-03-15:
If you remove the flag, I suspect you'll need to add a runtime request for `android.permission.ACCESS_FINE_LOCATION` instead. 

This can be done by using something like the cordova-diagnostic-plugin `requestRuntimePermissions` ([docs](https://github.com/dpa99c/cordova-diagnostic-plugin#requestruntimepermissions)) prior to starting the scan.