don/cordova-plugin-ble-central #944: Add method for permission state.

infacto, 2022-11-22:
Should we add a method for the new Android 12+ permissions? 

| Method | Description |
| --- | --- |
| `isEnabled` | Reports if bluetooth is enabled. |
| `isLocationEnabled` | Reports if location services are enabled. |
| `isScanEnabled` | Reports if bluetooth can scan. (Android 12, API 31) `new` |

How to test permission `BLUETOOTH_SCAN`? How to test if the bluetooth plugin is fully ready? 
Is it possible to test the permission before using it? 

My test: On Android 12 you only get the permission dialog two times. When you get the error "Bluetooth scan permission not granted." Btw. before this error the app event `platform.pause` is emitted and after this error `platform.resume`. That's weird. Another thing is, I have no idea where to enable BLUETOOTH_SCAN in Android settings. I cannot find it. Currently I have to uninstall the app and install again to trigger the permission dialog. We need a way to trigger this again or open settings if the user must manually enable it. 

Ref. #875

peitschie, 2022-11-22:
Hi @infacto, agree this would be a useful feature to have. 

In my current applications I've been using the [cordova-diagnostic-plugin](https://github.com/dpa99c/cordova-diagnostic-plugin) to support this workflow, but it's certainly a pain having to keep these permission checks aligned with the plugin as well, especially given the permissions are different for each Android release.

infacto, 2022-11-22:
Ok thanks, I found the permission now. Another issue is: You can disable location at any time. For example pull down the Android control center and tap on location. Then then call this ble plugin `startScanWithOptions` and see that no error appears. It behaves like just no device found.

peitschie, 2022-11-22:
Though most phones do require location services enabled to scan, not every phone does. I haven't yet figured out a good solution to tell whether the scan works or not, as Android doesn't return any error, it just ignored the scan request silently.

More discussion in this can be found here: https://github.com/don/cordova-plugin-ble-central/issues/872