don/cordova-plugin-ble-central #945: Different ID format

paulbarkhov-daw, 2022-11-25:
Peripheral Data is passed to the success callback when scanning and connecting. IOS gets ID in format like "15B4F1C5-C9C0-4441-BD9F-1A7ED8F7A365" and Android like "00:1A:7D:DA:71:13". How do I convert IOS ID format to Android ID format or visa versa?

HankLloydRight, 2022-11-25:
As far as I know, you can't 'convert' one format to the other... your code just needs to store/handle both ID formats. The Apple ID is a UUID which is 128 bits, and the Android ID looks like a MAC address which is 48 bits, so they're not interchangeable/convertible.

peitschie, 2022-11-28:
Hi @paulbarkhov-daw ... it is as @HankLloydRight has said. These IDs are generated by the operating system, and there's no method for mapping between these on different platforms.

If you need to have the MAC address accessible on iOS, I believe the only solution is to put it into the advertising data somehow, as iOS does not natively expose this field.

Related to #860 & #710

I'll close this out as there's little the plugin can do. However, feel free to continue the conversation here if you need!

closed 2022-11-28 by peitschie

paulbarkhov-daw, 2022-11-29:
@HankLloydRight @peitschie Any chance you guys know how to get this 128bit uuid suitable for iOS on ESP32?

HankLloydRight, 2022-11-29:
Can you explain what you mean by “suitable”?