don/cordova-plugin-ble-central #947: Getting strange ACCESS_FINE_LOCATION denial on Android 12, targeting API 31

QuentinFarizon, 2022-12-07:
I am on the latest version of this plugin (1.6.1-cli) and latest cordova-android (11.0.0), so targeting API 31

On some devices (including my personal phone, a Sony Xperia 1 II on Android 12), scanning doesn't work at all. It pormpts for permissions (I even use cordova.plugins.diagnostic to premptively prompt them), but scanning returns no results.

I have tried looking at logcat for some ideas, and I see a strange log about permissions :
`BluetoothUtils: Permission denial: Need ACCESS_FINE_LOCATION permission to get scan results`
From my understanding, on a Android 12 phone targeting API 31, this permission should not be needed at all.

peitschie, 2022-12-07:
Hi @QuentinFarizon ... it sounds like you've root-caused it correctly there! That's a bit unideal.

As a quick test, can you try asking for the `ACCESS_FINE_LOCATION` permission via the diagnostic plugin for this specific phone and see if that allows the scan results to be reported?

It does sound like the Sony Android 12 variant there is non-compliant with [Android](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android12-or-higher) (as you've already noted)

tigohenryschultz, 2023-01-06:
This plugin is breaking my Android app. If I don't use this plugin then my @capacitor/geolocation works fine. With this plugin installed my ACCESS_FINE_LOCATION / ACCESS_COARSE_LOCATION issues arise. I can't use this plugin right now.

tigohenryschultz, 2023-01-06:
Something like this may fix for Android 31/32

```
    if (targetSdkVersion >= 31) {
        // Strip out Android 12+ changes
        androidManifest = stripPermission(androidManifest, 'ACCESS_FINE_LOCATION');
        androidManifest = stripPermission(androidManifest, 'ACCESS_COARSE_LOCATION');
        androidManifest = stripPermission(androidManifest, 'ACCESS_BACKGROUND_LOCATION');
        manifestChanged = true;
    }
```

peitschie, 2023-01-06:
hi @tigohenryschultz , it sounds like you're hitting #932 

The fix here is to switch to the slim variant of this plugin, and manually control those permissions yourself, as per https://github.com/don/cordova-plugin-ble-central/issues/932#issuecomment-1307057842

If that doesn't work, please feel free to raise a new issue.

tigohenryschultz, 2023-01-06:
Okay I will try that on Monday for now I built my app on a fork with my changes:

https://github.com/tigohenryschultz/cordova-plugin-ble-central

peitschie, 2023-01-07:
The specific issue you're facing there @tigohenryschultz  is that this plugin now specifies a maxSdk version for coarse & fine location, which messes up Cordova's somewhat naive XML merge logic. What your workaround there is doing is simply removing one of the offending permission entries... but it's unfortunately non-deterministic in my experience, so was not suitable to build into the plugin itself 😩 .

The only solution in the end is to remove those permission entries completely, as that is the only way to ensure no conflicts with other plugins.