don/cordova-plugin-ble-central #949: Read failed - Medical Device BLE

softwareenginesrl, 2022-12-29:
Hello, when I try to connect to my multiparameter medical device I get the error in object. The device requires the use of a pairing PIN. I have developed everything in Android. Do you have any ideas on how I can solve it? Thanks

**CONNECTION**
```

connect: function(e) {
        var deviceId = e.target.dataset.deviceId,
            onConnect = function() {
                batteryStateButton.dataset.deviceId = deviceId;
                disconnectButton.dataset.deviceId = deviceId;
                app.showDetailPage();
            };

        ble.connect(deviceId, onConnect, app.onError);
        ble.setPin(3618,
        function successo(data){
        console.log('#pin_success', JSON.stringify(data))
        }, function errore(data){
        console.log('#pin_error', JSON.stringify(data))
        });
    },
```

**READ**

```
readBatteryState: function(event) {
        console.log("readBatteryState");
        alert('provo a leggere');
        var deviceId = event.target.dataset.deviceId;
        var service_uuids="0899b07e-7489-b7a0-7781-a405d2675cb9";
 var characteristic_uuid= "0222b02e-8232-b2a0-4111-a222d2234cb5"; //Read Failed

        ble.read(deviceId,service_uuids,characteristic_uuid,function(data) {
        console.log('#info', JSON.stringify(data))
       }, function(data) {console.log('#info','errore',data)});

    },
```

peitschie, 2023-01-06:
Hi @softwareenginesrl 

What are the logs you get to the console based on the above code?

peitschie, 2023-04-21:
Closing this as stale. Please feel re-open with further information to aid with debugging any potential issue.

closed 2023-04-21 by peitschie