don/cordova-plugin-ble-central #950: Do this plugin can works with BLE door lock?

scramatte, 2023-02-07:
Hi,

Do this plugin can works with BLE door lock?

Regards

peitschie, 2023-02-07:
Hi @scramatte 

It ought to be possible, but the plugin itself is not enough to control the door. You still need to figure out the right services and characteristics to interact with in order to achieve what you're wanting to there.

If you use nrf Connect or a similar mobile device, you should be able to scan for the door locally and connect to it and test out some things.

scramatte, 2023-02-07:
Software of the lock is crap and SDK in opinion is very expensive for what is it.  
I'm quite new in this. Do you know if it possible to sniff  traffic in some way? 
Does it possible to hire your service to make some tests?

peitschie, 2023-02-07:
@scramatte it's likely to be difficult to reverse engineer, as I'm going to assume that the lock has some kind of security in place (hopefully... otherwise it's not the kind of lock you want!)

Does the company produce any kind of technical documentation on how to interact with their locks? (Most consumer products do not, but it's worth looking).

Traffic sniffing is possible with Wireshark and an Android phone: https://stackoverflow.com/a/23900782/156169

But... it's not trivial to figure out what's from those logs.

Regarding my services, I don't have a lot of spare capacity to help at this time, sorry 😅

peitschie, 2023-03-16:
I'm closing this out as there's no issue or missing feature in the plugin here. Feel free to continue the discussion on the ticket however!

closed 2023-03-16 by peitschie