don/cordova-plugin-ble-central #961: Scan() - local name not returned with iOS 16.4.1

ASHBAW, 2023-04-25:
I have noticed that with the new iOS 16.4.1 that local name is no longer returned:

```
2023-04-25 11:07:40.113674-0500 App[716:54224] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataServiceUUIDs =         (
            "832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1"
        );
        kCBAdvDataTxPowerLevel = "-2";
    };
    id = "DBC2205A-AB82-D6A5-6D71-4CB132615472";
    name = "Hickory    ";
    rssi = "-90";
    state = disconnected;
}
```

[image]

With an older version of iOS **kCBAdvDataLocalName** is returned

```
2023-04-25 11:15:55.734409-0500 App[320:5283] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = test1;
        kCBAdvDataServiceUUIDs =         (
            "832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1"
        );
        kCBAdvDataTxPowerLevel = "-2";
    };
    id = "29C7EF76-9CAA-6B8B-A14A-AFA1045A69AF";
    name = "Hickory    ";
    rssi = "-85";
    state = disconnected;
}
```

[image]

If I use a generic Bluetooth app and connect to my device I see a local name (serial number of our device in this case):

[image]

Is there anything that can be done to get this functionality back? We currently use the local name in app's scan list to differentiate between multiples of our device.

Thanks,

Brent

peitschie, 2023-04-26:
Hi @ASHBAW 

Interesting issue you're seeing there! What scan options are you using to initiate the scan?

Unfortunately, I'm not seeing this same issue with iOS 16.4.1:
```
2023-04-26 11:39:14.295375+1000 App[586:22398] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = TEMPT;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "//8BAgME";
        };
        kCBAdvDataServiceUUIDs =         (
            "4A1C3000-ED6B-47DB-8EC7-22D9A44FE681"
        );
    };
    id = "702FFC4F-B44B-8D62-39CE-3C225208FD35";
    name = TEMPT;
    rssi = "-62";
    state = disconnected;
}
```

Do you know if this data is being included in the primary advertising packet or the scan response packet? For my own local testing, I'm using the primary advertising packet.

If you haven't already tried it, does the behaviour change if you try [ble.startScanWithOptions](https://github.com/don/cordova-plugin-ble-central#startscanwithoptions) with `reportDuplicates` set to true?
E.g.,
```
ble.startScanWithOptions([], { reportDuplicates: true}, console.log, console.error);
```

ASHBAW, 2023-04-26:
The data is being included in the scan response packet.

And thank you for your suggestions. When I use your suggestion I do see the local name:
```

2023-04-26 08:03:00.841236-0500 App[467:27844] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = test1;
        kCBAdvDataServiceUUIDs =         (
            "832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1"
        );
        kCBAdvDataTxPowerLevel = "-2";
    };
    id = "DBC2205A-AB82-D6A5-6D71-4CB132615472";
    name = "Hickory    ";
    rssi = "-82";
    state = disconnected;
}

```

I am trying to understand why this works?

Thanks again for you help.

ASHBAW, 2023-04-26:
Looking at it more closely the first response does not have the local name in it but the next one does. Time stamps indicates this happening back-to-back.

Any insight on this?

```
2023-04-26 08:26:05.437221-0500 App[582:38042] Discovered {
    advertising = {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataServiceUUIDs =   (
            "832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1"
        );
        kCBAdvDataTxPowerLevel = "-2";
    };
    id = "1EA62759-D381-4FBC-E314-A89B1535326E";
    name = "Hickory    ";
    rssi = "-86";
    state = disconnected;
}
```

```
2023-04-26 08:26:05.437852-0500 App[582:38042] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = 000000000;
        kCBAdvDataServiceUUIDs =         (=
            "832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1"
        );
        kCBAdvDataTxPowerLevel = "-2";
    };
    id = "1EA62759-D381-4FBC-E314-A89B1535326E";
    name = "Hickory    ";
    rssi = "-86";
    state = disconnected;
}
```

peitschie, 2023-04-27:
When report duplicates is disabled, iOS collates multiple advertising packets together and raises an event which includes all this information at once: https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerscanoptionallowduplicateskey

It's possible that a recent update (either in the OS, or in the build tools) has tweaked this behaviour slightly, and perhaps the advertising packets are no longer being stitched together the same way.

But... it's just a guess! I think the comment here on the capacitor BLE plugin is likely correct: https://github.com/capacitor-community/bluetooth-le/issues/528#issuecomment-1523753073

peitschie, 2023-04-27:
Just collecting together other related issues from a similar code-base
* https://github.com/capacitor-community/bluetooth-le/issues/527
* https://github.com/capacitor-community/bluetooth-le/issues/528