don/cordova-plugin-ble-central #964: How to prevent addition of android.permission.ACCESS_FINE_LOCATION

mikerusso-tsi, 2023-05-03:
What is the proper means to prevent the plugin from adding android.permission.ACCESS_FINE_LOCATION to AndroidManifest.xml?

graphefruit, 2023-05-03:
Did you have a look here: https://github.com/don/cordova-plugin-ble-central/issues/932#issuecomment-1307057842 ?

mikerusso-tsi, 2023-05-03:
Okay, thanks!    The comment I saw there from Philip a little later in that issue says it all:
https://github.com/don/cordova-plugin-ble-central/issues/932#issuecomment-1311025592
I.e., there's no mechanism to prevent the plugin from adding a permission other than to use the @slim version of the plugin that does not add any permissions in the first place!

I tried this but I was foiled by Android: I installed the @slim plugin and then edited my config.xml to add all permissions indicated on the plugin's page except for ACCESS_FINE_LOCATION (and BLUETOOTH_CONNECT since app does not connect to already-paired devices).
The result was that when I started the app, Android never presented a location permission dialog, and all ble.scan()'s failed for lack of it.  This was on a Version 10 tablet.  I uninstalled the app beforehand and tried to make sure that a previous permission denial wasn't lurking.

So, I had to leave FINE in there (along with both COARSE) and used the android:usesPermissionFlags="neverForLocation" attribute on the BLUETOOTH_SCAN to "strongly assert that my app doesn't derive physical location" to avoid unnecessary scrutiny.

mikerusso-tsi, 2023-05-03:
With regards to my "foiled by Android" remark above (the location permission dialog was never presented when only the COARSE was in config.xml), perhaps it had something to do with the android.json file... many of the permission entries had count fields of 10 for some reason.  I deleted the file and did another build and the counts were then 1 as I would expected.

peitschie, 2023-05-03:
`ACCESS_FINE_LOCATION` is mandatory for scanning in Android 10, see https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android11-or-lower

You'll notice the plugin's default configuration follows the recommendation and puts a `maxSdkVersion="30"` on these legacy permissions so that they are not incorrectly requested on Android 12: https://github.com/don/cordova-plugin-ble-central/blob/b894ea07b38b181a818faef6e783f44427f4b14a/plugin.xml#L68-L75

But it sounds like you're on the right path. If you don't need connectivity, I agree the `slim` variant with your own config is the right way to go 🙂 

I'll close this out as it seems like you've solved your issue. Feel free to re-open though if the issue is persisting.

closed 2023-05-03 by peitschie

mikerusso-tsi, 2023-05-09:
> perhaps it had something to do with the android.json file... many of the permission entries had count fields of 10 for some reason.

I'm wondering if this is normal? After several builds, my android.json 'use-permission' counts are way up again... 22+.  Thanks!

peitschie, 2023-05-09:
@mikerusso-tsi what is the `android.json` file you are referring to here? Do you mean `AndroidManifest.xml`?

Are you able to paste the `use-permission` content here so I can have a peek?

The above snippet is 100% supported cordova syntax, so in theory cordova should have no issues with it 🤔 

Does this cause build problems for you?

mikerusso-tsi, 2023-05-09:
android.json is created during a build and located here under the project directory:
platforms/android/android.json

I believe it is used during generation of the AndroidManifest.xml.  It hasn't caused me problems yet TMK but it seems very suspicious that the counts keep incrementing.

Here is the section I am referring to:
`  "config_munge": {
    "files": {
      "res/xml/config.xml": {
        "parents": {
          "/*": [
          ],
          "/widget": [
            {
              "xml": "<feature name=\"BLE\"><param name=\"android-package\" value=\"com.megster.cordova.ble.central.BLECentralPlugin\" /></feature>",
              "count": 1
            }
          ]
        }
      },
      "AndroidManifest.xml": {
        "parents": {
          "/manifest": [
            {
              "xml": "<uses-permission android:name=\"android.permission.VIBRATE\" />",
              "count": 1
            },
            {
              "xml": "<uses-permission android:maxSdkVersion=\"28\" android:name=\"android.permission.ACCESS_COARSE_LOCATION\" />",
              "count": 28
            },
            {
              "xml": "<uses-permission android:maxSdkVersion=\"30\" android:name=\"android.permission.BLUETOOTH\" />",
              "count": 28
            },
            {
              "xml": "<uses-permission android:maxSdkVersion=\"30\" android:name=\"android.permission.BLUETOOTH_ADMIN\" />",
              "count": 28
            },
            {
              "xml": "<uses-permission android:name=\"android.permission.BLUETOOTH_SCAN\" android:usesPermissionFlags=\"neverForLocation\" />",
              "count": 26
            },
            {
              "xml": "<uses-permission android:name=\"android.permission.BLUETOOTH_CONNECT\" />",
              "count": 3
            },
            {
              "xml": "<uses-permission android:maxSdkVersion=\"30\" android:name=\"android.permission.ACCESS_FINE_LOCATION\" />",
              "count": 20
            }
          ],
          "/*": [
            {
              "xml": "<uses-permission android:name=\"android.permission.WRITE_EXTERNAL_STORAGE\" />",
              "count": 1
            }
          ]
        }
      }
    }
  },
`

peitschie, 2023-05-09:
Gotchya!

What does that look like after a few runs when you're seeing the 22 entries? Also, what other plugins are you using here?

peitschie, 2023-05-09:
Oh, right, you're referring to the `"count"` section there?

The long and short is this is cordova's business 🙂 

I do agree it looks a bit funny, but the plugin itself has no direct control over this. The `plugin.xml` here adheres with Cordova's published specification: https://cordova.apache.org/docs/en/latest/plugin_ref/spec.html#config-file

If there is any actual bug, it's likely to be on cordova's side of the fence.

(just confirming too, this is happening with the `slim` variant or the non-slim one?)

mikerusso-tsi, 2023-05-09:
This was while using the non-slim (the PR code).  But I did leave my subset of the bluetooth permissions in my config.xml.

However, when I originally mentioned this way above, it was with the slim variant.