don/cordova-plugin-ble-central #966: `cordova plugin add` doesn't add `<plugin>` attribute to `config.xml`

laralove143, 2023-05-05:
I've tried this a few times even on fresh projects, it just won't add the required line `<plugin name="cordova-plugin-ble-central" spec="~2.1.0" />` to `config.xml`, it does everything else like variables (used for iOS for example) and the node package. If it's not supposed to add it, it'd be nice to include that we have to add the line under the _Installation_ heading

peitschie, 2023-05-05:
The addition of the plugin element to the config.xml is all controlled by the Cordova CLI tool.

This element hasn't been needed or used since v6.1 according to the Cordova docs: https://cordova.apache.org/docs/en/latest/reference/cordova-cli/index.html#algorithm-for-resolving-plugins

What you are describing wouldn't be needed, as there should be the proper entry added to package.json automatically by Cordova.

closed 2023-05-05 by peitschie

laralove143, 2023-05-06:
without adding this, i got an error `ble` variable not defined

peitschie, 2023-05-06:
What version of Cordova are you using?

peitschie, 2023-05-06:
Out of interest, what do you see in the config file when you add one of the core plugins such as cordova-plugin-network-information ?

Installation instructions as per here:  https://cordova.apache.org/docs/en/latest/reference/cordova-plugin-network-information/index.html#installation

laralove143, 2023-05-08:
command log:
```% cordova plugin add cordova-plugin-network-information
Installing "cordova-plugin-network-information" for browser
Installing "cordova-plugin-network-information" for ios
Adding cordova-plugin-network-information to package.json
```
`config.xml` is not changed

peitschie, 2023-05-09:
`package.json` is all that's required. No manual modified of `config.xml` is needed for any reason 🙂

laralove143, 2023-05-10:
In my case that didn't work, I couldn't access the `ble` variable even though I followed the examples 1:1

peitschie, 2023-05-10:
@laralove143 do you have any cordova elements installed globally? I you can see all the npm global scripts via `npm list --global`

If so, can you confirm which versions are there?

All I keep coming back to is that if the `config.xml` changes are needed... somehow you have elements of an older cordova version involved somewhere in your build chain.

Have you tried deleting your plugins and platforms folders, and rebuilding them again from scratch via `cordova prepare android` and `cordova prepare ios`?

laralove143, 2023-05-12:
I don't think I do:
```
lara@Laras-MacBook-Air SpoonLocal % npm list --global  
/usr/local/lib
├── cordova@11.1.0
├── npm@9.6.4
└── prettier@2.8.8
```

I don't think I have any other plugins to begin with:
```
lara@Laras-MacBook-Air SpoonLocal % npm list
io.cordova.hellocordova@1.0.0 /Users/lara/WebstormProjects/SpoonLocal
├── cordova-browser@6.0.0
├── cordova-ios@6.3.0
├── cordova-plugin-ble-central@1.6.3
├── cordova-plugin-network-information@3.0.0
└── ios@0.0.1
```
I don't remember adding `cordova-plugin-network-information` maybe it's an indirect dependency?

peitschie, 2023-05-12:
Can you paste the full content of your config.xml?

laralove143, 2023-05-15:
```xml
<?xml version='1.0' encoding='utf-8'?>
<widget id="io.cordova.spoonlocal" version="1.0.0" xmlns="http://www.w3.org/ns/widgets"
        xmlns:cdv="http://cordova.apache.org/ns/1.0">
    <name>SpoonLocal</name>
    <description>Sample Apache Cordova App</description>
    <author email="dev@cordova.apache.org" href="https://cordova.apache.org">
        Apache Cordova Team
    </author>
    <content src="index.html"/>
    <access origin="*" />
    <allow-intent href="http://*/*"/>
    <allow-intent href="https://*/*"/>

    <platform name="ios">
        <config-file parent="UIBackgroundModes" target="*-Info.plist">
            <array>
                <string>bluetooth-central</string>
            </array>
        </config-file>
    </platform>

    <plugin name="cordova-plugin-ble-central" spec="~2.1.0" />
</widget>
```
this is just the template that i haven't worked on customizing yet

peitschie, 2023-05-16:
@laralove143 this really has me stumped.

Just for comparison, I've pushed up my own clean cordova app based off the same templates you're using here: https://github.com/peitschie/cordova-ble-simple-app

Is it worth pulling that down and seeing if you get any different behaviour with that app?

[image]

laralove143, 2023-05-16:
I'm unable to do `npm install` for that repo since https://artifactory.pi.planetinnovation.com.au seems to be down, maybe that's temporary

peitschie, 2023-05-17:
@laralove143 that wasn't good planning on my part 😆 

Should be fixed now... try again?

laralove143, 2023-05-20:
still the same :( heres the exact output
```
lara@Laras-MacBook-Air cordova-ble-simple-app % npm i
npm WARN old lockfile 
npm WARN old lockfile The package-lock.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile 
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile 
npm ERR! code ENOTFOUND
npm ERR! syscall getaddrinfo
npm ERR! errno ENOTFOUND
npm ERR! network request to https://artifactory.pi.planetinnovation.com.au/artifactory/api/npm/planet-npm/@nodelib/fs.walk/-/fs.walk-1.2.8.tgz failed, reason: getaddrinfo ENOTFOUND artifactory.pi.planetinnovation.com.au
npm ERR! network This is a problem related to network connectivity.
npm ERR! network In most cases you are behind a proxy or have bad network settings.
npm ERR! network 
npm ERR! network If you are behind a proxy, please make sure that the
npm ERR! network 'proxy' config is set properly.  See: 'npm help config'

npm ERR! A complete log of this run can be found in: /Users/lara/.npm/_logs/2023-05-20T20_26_21_134Z-debug-0.log
lara@Laras-MacBook-Air cordova-ble-simple-app % 
```

peitschie, 2023-05-20:
Try deleting `package-lock.json` and just doing `npm install`

laralove143, 2023-05-21:
sorry for the hassle, i did it and it works, i also went back to my repo and deleted the line from `config.xml` and it still works but maybe im supposed to rebuild or something to clean the incremental artifacts? or maybe i've just been mysterious bugged

peitschie, 2023-05-21:
@laralove143 no worries at all! Glad you got sorted out.

I occasionally find myself needing to delete  the `plugins` and `platforms` after certain activities to make everything regenerate properly. I find it's good practice to delete both these folders after doing an `npm ci` or `npm install`, as cordova makes links to installed versions of the NPM packages which may be broken when the `node_modules` folder is removed.