don/cordova-plugin-ble-central #971: Getting error when reading data from device BLE-9909 Meter Powered by YINMIK Mobile App

phucluke, 2023-05-18:
Hi team,

I'm building an cordova app for reading data from device BLE-9909 Meter Powered by YINMIK.
The scanning and connecting with plugin are fine except the reading data. I'm getting the following error. Could you please provide a direction? Thanks.
```
D/BLEPlugin: action = isConnected
D/BLEPlugin: action = read
D/Peripheral: Queuing Command com.megster.cordova.ble.central.BLECommand@ca43bed
D/Peripheral: Processing Commands
D/Peripheral: Read 0000ff02-0000-1000-8000-00805f9b34fb
D/Peripheral: onCharacteristicRead android.bluetooth.BluetoothGattCharacteristic@cb961e5
D/Peripheral: Processing Complete
D/Peripheral: Processing Commands
D/Peripheral: Command Queue is empty.

I/chromium: [INFO:CONSOLE(478)] "Error reading 0000ff02-0000-1000-8000-00805f9b34fb status=133"
```

Here is the code snipped:
```
ble.startNotification(
  deviceId,
  'FF01',
  'FF02',
  (buffer) => {
    this.decodeDataFromBLE9909Sensor(buffer, callback);
  },
  (err) => {
    console.error(err);
  }
);
```
And device document
```
Bluetooth communication protocol												

Basic Information												
1	Bluetooth name:	BLE-9100	i.e. product name(BLE-9100,BLE-Ph01,BLE-9909,…)									
2	Service UUID	FF01										
3	Feature UUID	FF02	read and write									

read data												
1	After the Bluetooth connection is successful, subscribe to the characteristic UUID FF02,											
2	Read the characteristic UUID FF02 to return the data											

```

peitschie, 2023-05-19:
@phucluke are you able to see if you can read this service/characteristic with [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) on your Android phone?

From the adb logs, there should be a bunch more data in theory about the underlying BLE protocol between those entries you've logged there. Are you able to extra this and include it also?

I guess the last fairly normal question is does this happen across all android versions/devices, or only specific ones?

phucluke, 2023-05-19:
Hi @peitschie
Thanks for your support.
with nrf Connect, I was able to scan and connect the service/characteristic.
Here is the screenshot:
[image]
The value needs to be decoded to show but what we need is to get the data first.
We tested on some kind of android devices and got the same issue.

phucluke, 2023-05-20:
Hi @peitschie 
I got nothing when clicking on the down arrow on nrf Connet but we could get data with [Bluefruit Connect](https://play.google.com/store/apps/details?id=com.adafruit.bluefruit.le.connect&hl=en)
[image]

[image]

peitschie, 2023-05-20:
@phucluke can you try using this characteristic UUID instead? `ADAFFF02-C332-42A8-93BD-25E905756CB8`

Based on adafruit's documentation, it seems their 16-bit UUID is not a standard Bluetooth one: https://github.com/adafruit/bluetooth-low-energy#adafruit

Just to confirm this, in the `connect` callback you should receive a device info object that has a list of all the services and characteristics discovered on the device. Can you paste that here as well?

phucluke, 2023-05-20:
Hi @peitschie,
I tried but getting similar error: `Characteristic adafff02-c332-42a8-93bd-25e905756cb8 not found`
I also tried to build and install the app for an iOS device and could get the data. It happened only on Android.
Here is the peripheral object got in the connect callback:
```
{
  "name": "BLE-9909",
  "id": "C0:00:00:00:51:40",
  "advertising": {},
  "rssi": -80,
  "services": [
    "1800",
    "1801",
    "180a",
    "180f",
    "ff01"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a23",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a24",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a25",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a26",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a27",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a28",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a29",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a2a",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a50",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180f",
      "characteristic": "2a19",
      "properties": [
        "Read",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2908"
        }
      ]
    },
    {
      "service": "ff01",
      "characteristic": "ff02",
      "properties": [
        "Read",
        "Write",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2901"
        }
      ]
    },
    {
      "service": "ff01",
      "characteristic": "ff10",
      "properties": [
        "Read"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        }
      ]
    }
  ]
}
```

peitschie, 2023-06-01:
If you're able to help with debugging a bit, we might be able to dig a bit more.

That error is thrown here: https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L775C54-L781

Are you able to get Android studio attached and drop a breakpoint there so you step through `findWritableCharacteristic`?

I wonder if the issue is that we are looking for the expanded `characteristicUUID`, but the service discovery only sees the short one here https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L817

If you can help me figure out what `getUuid` returns for the `ff02` characteristic there, I can probably figure out the code changes to make this function.

phucluke, 2023-06-08:
Hi @peitschie 
I very much appreciate your support.
I was debugging on `findReadableCharacteristic` and it was fine to get the characteristic
[image]

[image]

And I found the exception here
[image]

vanminh0910, 2023-06-18:
I experienced the same issue and jumped to this thread from Internet search.

Great if we can have any solution for it. Thanks.

peitschie, 2023-06-19:
@phucluke unfortunately I've got no great insights here 🤔 

The fact that this fails in both this plugin and nrf Connect means it's more likely a quirk in the peripheral, but I'm stumped as to why Bluefruit functions better here.

Bluefruit's own code here seems to be following exactly the same pattern as this plugin: https://github.com/adafruit/Bluefruit_LE_Connect_Android_V2/blob/master/app/src/main/java/com/adafruit/bluefruit/le/connect/ble/central/BlePeripheral.java#L706

Unfortunately, I don't have any access to a local device to try anything out myself.

Are there any support forums within the adafruit community that might have some ideas?