don/cordova-plugin-ble-central #972: ios 16.5 support issue

jihen-ben-said, 2023-05-25:
In my Cordova application for iOS, I have encountered a peculiar issue. The application works flawlessly on devices such as the iPhone Xs running iOS 16.5. However, when testing it on the iPhone 13 with iOS 16.5, I noticed a problem. After initiating the startNotification process, the data received appears to be corrupted.

peitschie, 2023-05-25:
Hi @jihen-ben-said 

That certainly sounds surprising!

Are you able to share a good and bad notification example, just to see what kind of corruption you're dealing with? Is it a partial packet? Is it random bytes? Too long? Too short? Reversed?

Apple is usually pretty good about keeping their iOS behaviour consistent between hardware phones, but I have seen the occasional bug slip through.

jihen-ben-said, 2023-05-30:
[image]
[image]
 Hi,
In the following screenshots, one contains successfully received data, and the other contains yellow, indicating that a brace is missing, so invalid data. This is a random problem, as sometimes other characters such as "a/z/f",}..." or a number are missing.

peitschie, 2023-06-01:
@jihen-ben-said do you have any control over the peripheral here? Do you know if this is using a Bluetooth **notification** or an **indication**?

The reason this matters, is because notifications are not guaranteed delivery and allowed to be dropped or omitted by the BLE protocol (e.g., like a UDP packet). You could possibly check this by using [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to connect to device, and look whether the `INDICATE` flag is set for this characteristic, or just the `NOTIFY` flag.

This could explain why the data goes missing, and only goes missing sometimes.

jihen-ben-said, 2023-06-15:
Hi,
the notify characteristic have indicate true but the writing characteristic have indicate false