don/cordova-plugin-ble-central #976: Scan result - No error but totally blank

clarklight, 2023-06-06:
I used the @slim version and have the permissions below, when i copied and pasted in the demo code to test out a scan, it says scan completed, but returned no item, but i have atleast a few bluetooth device next to me??? Any pointer? 

Thanks
Clark 

```
	<config-file target="AndroidManifest.xml" parent="/manifest">
		<uses-permission android:name="android.permission.CAMERA" android:maxSdkVersion="30" />
		<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
		<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
		<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
		<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
		<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
		<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
		<uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />
		<uses-permission android:name="android.permission.RECORD_AUDIO" />

	</config-file>
```

clarklight, 2023-06-06:
@peitschie Thank you so much for the quick pointer!! I managed to get it to work, i did have Location setting, but i didnt know that it won't give me a prompt if it isnt turned on, i just turned the Location on, and it is showing the devices now!! 
Thank you so much!
Clark

peitschie, 2023-06-07:
Fantastic. Glad it was resolved so readily.

Just as a bit of background location services are not required on all phones or Android versions to scan... so it's a bit hit and miss as to whether it's needed or not. Hence, the plugin itself doesn't reject a scan even when location services are disabled because we have no way of knowing if the scan works or not.

There are APIs available to listen for updates to the location services if you need them in your app: https://github.com/don/cordova-plugin-ble-central#islocationenabled

closed 2023-06-07 by peitschie