don/cordova-plugin-ble-central #979: Some BLE not found

valeprog, 2023-06-15:
I have been using the plugin for some years without problems, now since I updated to the latest version with the scan I no longer see some BLEs, while if I reinstall my previous app I find them correctly

peitschie, 2023-06-15:
Hi @valeprog 

Android 12+ has some documentation suggesting BLE beacons are filtered from scan results if one of these are missing: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#assert-never-for-location

When targeting android 12+, you'll need to use the slim variant of the plugin ([instructions](https://github.com/don/cordova-plugin-ble-central#android-permission-conflicts)) and remove the `android:usesPermissionFlags="neverForLocation"` bit from the example manifest. Then you'll need to request ACCESS_FINE_LOCATION on Android 12 at runtime. I usually use the cordova.plugin.diagnostics for this.

Unfortunately, there's no way for me to wrap the above up in the plugin, but I might put together some better documentation about how to do this with an example.

valeprog, 2023-06-16:
Hi @peitschie 
Ok thanks I'll try.

valeprog, 2023-06-19:
ok without android:usesPermissionFlags="neverForLocation" it work fine.

peitschie, 2023-06-19:
Fantastic. Glad you were able to sort things out @valeprog

closed 2023-06-19 by peitschie