don/cordova-plugin-ble-central #986: Gatt Callback issue

Jatindersyall12, 2023-08-16:
@don

I am facing connection callback issue. I want to connect two BLE devices with GATT server but below code say's it's already connected, but my devices never connected. Can you please help me out on this.

               bluetoothGatt = device.connectGatt(
                this,
               true,
               object : BluetoothGattCallback() {
               override fun onConnectionStateChange(gatt: BluetoothGatt, status: Int, newState: Int) {
        
                when (newState) {
                    BluetoothProfile.STATE_CONNECTED -> {
                        gatt.discoverServices()
                    }
                    BluetoothProfile.STATE_DISCONNECTED -> {

                        //deviceStatusListener?.onDisconnect()
                        //closeConnection()
                    }
                }
Looking for your's valuable feedback soon.

Thanks Jatinder Syall

peitschie, 2023-08-17:
Hi @Jatindersyall12 

That code there does not look like anything related to this plugin? This is a cordova/javascript wrapper around Bluetooth... what you've posted there looks like kotlin.

If you're looking for native Android Bluetooth development support, I'm afraid this is a long way beyond the plugin scope here 🙂 

Given this disclaimer... the code you have there looks ok, but it's hard to guess because this is such a small piece of the code, and I have no real idea what you're doing around this snippet.

> ... below code say's it's already connected, but my devices never connected

What do you mean by "it's already connected"? As in, the `BluetoothProfile.STATE_CONNECTED` call is triggered, or as in the `connectGatt` method completes and execution continues? How do you know the devices never connect?

peitschie, 2023-08-17:
I'm closing this as it's not a plugin issue at all... but feel free to continue the discussion here. I'll reply if I'm able to with general tips, but it might be worth reading through the native Android plugin code here to try and get a feel for how those callbacks should behave.

There are tons of native Android BLE connection tutorials out there that could hopefully get you moving along.

Check out https://punchthrough.com/android-ble-guide/ for the most complete one I've found.

closed 2023-08-17 by peitschie