don/cordova-plugin-ble-central #988: Accu Chek Mobile with Bluetooth adaptator

BenLaKnet, 2023-08-31:
Hi,

After many months, all is running, but now, I do not understand why I can not write to BLE.
I have a write error message 129, when I am doing this:

`var databuffer = new Uint8Array(2);
	databuffer[0] = 0x01;
	databuffer[1] = 0x01;
		
	ble.startNotification(localStorage.getItem('ACMid'), "1808", "2A18", 
		function(datanotif){
			
			int8data = new Uint8Array(datanotif);			
			//console.log("Hooray we have data notif : " + JSON.stringify( int8data ));
						
			syncid = int8data[1] & 255;
			syncid |= (int8data[2] & 255) << 8;
						
			val = int8data[12];
					
			annee = int8data[3] & 255;
			annee |= (int8data[4] & 255) << 8;

			mois = addZero(int8data[5]);
			jour = addZero(int8data[6]);
			
			h = addZero(int8data[7]);
			m = addZero(int8data[8]);
			s = addZero(int8data[9]);
			
			datehms = annee + "-" + mois + "-" + jour + " " + h +":"+m+":"+s;
			
			//console.log("syncid: " + syncid + " val: " + val + " à " + datehms);
			
			if (! arrFait[syncid]) {
				
				arrFait[syncid] = {
					syncid	: syncid,
					val		: val,
					datehms	: datehms
				}
			}
			
		},
		function(failure){
			addMyContenu("Failed notif from device. " + JSON.stringify(failure), 1);
			console.log("Failed notif from device. " + JSON.stringify(failure));
		});		
	
	ble.write(localStorage.getItem('ACMid'), "1808", "2A52", databuffer.buffer, 
		function(data){
			console.log("Hooray we have data write : " + JSON.stringify( data ));				
		}, 
		function(failure){
			addMyContenu("Failed write from device. " + JSON.stringify(failure), 1);
			console.log("Failed write from device. " + JSON.stringify(failure));
		});`

Please... help...

And I have problems with iOS, I do not understand why it does not work, when I scan peripheral...

peitschie, 2023-08-31:
Hi @BenLaKnet 

Are you able to grab the ADB logs for one of these failures? E.g., `adb logcat > debug.logs`

Have you tried with [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to see if that gets the same issue on write?

It's pretty likely something specific with your device... but perhaps we can figure out what to do to workaround it...

BenLaKnet, 2023-09-01:
Many Thanks @peitschie ,

I tried your code update, I have to add : async function connectACMglycemiemeasurement() in my function.

But now I have this message:

""Uncaught (in promise) 129"

peitschie, 2023-09-01:
@BenLaKnet it's pretty hard to guess, as that log snippet there is very incomplete 🙂 

Have you tried writing with another application of some description?

BenLaKnet, 2023-09-01:
I will try with a little part of code into a new app and try with iOS too.
Many thanks @peitschie

BenLaKnet, 2023-09-02:
I have no entry in ble.withPromises.startNotification
I do not understand why it does not called.

Security ?

BenLaKnet, 2023-09-06:
@peitschie, have you changed something recently in your plugin ?
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/issues/89

I do not know when I updated my plugins.

peitschie, 2023-09-06:
@BenLaKnet I've made no changes for a long time in this area. I don't use the nRF Toolbox as part of this plugin.

> I have no entry in ble.withPromises.startNotification

What version of the plugin are you on currently? It might be worth trying 1.7.0 with the `await` syntax to see if that makes a difference?

Just to confirm, you are only starting notifications once after connecting to the device, correct? Not once per write?

BenLaKnet, 2023-09-06:
Yes , I am using 1.7.0
No changes with await.

It is difficult to use nrf Connect, because it is necessary to startNotification and write to obtain values.

`function startReadDatasConfirm() {
	
	arrFait = [];
	
	document.getElementById("BTNstartReadDatas").disabled=true;	
	addMyContenu("Reading...", 1);
	
	ble.connect(localStorage.getItem('ACMid'), connectACMglycemiemeasurement, connectACMfails);

}

async function connectACMglycemiemeasurement() {
	
	console.log("connectACMglycemiemeasurement");
	
	var databuffer = new Uint8Array(2);
	databuffer[0] = 0x01;
	databuffer[1] = 0x01;

		
	await ble.withPromises.startNotification(localStorage.getItem('ACMid'), "1808", "2A18", 
		function(datanotif){
			
			console.log("dedans Notif");
			
			int8data = new Uint8Array(datanotif);			
			//console.log("Hooray we have data notif : " + JSON.stringify( int8data ));
						
			syncid = int8data[1] & 255;
			syncid |= (int8data[2] & 255) << 8;
						
			val = int8data[12];
					
			annee = int8data[3] & 255;
			annee |= (int8data[4] & 255) << 8;

			mois = addZero(int8data[5]);
			jour = addZero(int8data[6]);
			
			h = addZero(int8data[7]);
			m = addZero(int8data[8]);
			s = addZero(int8data[9]);
			
			datehms = annee + "-" + mois + "-" + jour + " " + h +":"+m+":"+s;
			
			//console.log("syncid: " + syncid + " val: " + val + " à " + datehms);
			
			if (! arrFait[syncid]) {
				
				arrFait[syncid] = {
					syncid	: syncid,
					val		: val,
					datehms	: datehms
				}
			}
			
		},
		function(failure){
			addMyContenu("Failed notif from device. " + JSON.stringify(failure), 1);
			console.log("Failed notif from device. " + JSON.stringify(failure));
		});

	
	await ble.withPromises.write(localStorage.getItem('ACMid'), "1808", "2A52", databuffer.buffer, 
		function(data){
			console.log("Hooray we have data write : " + JSON.stringify( data ));				
		}, 
		function(failure){
			addMyContenu("Failed write from device. " + JSON.stringify(failure), 1);
			console.log("Failed write from device. " + JSON.stringify(failure));
		});
		
}`

I think the problem is with Accu Chek Mobile but I do not know how to reset connections
I tried to press button 5 seconds but nothing works.

BenLaKnet, 2023-09-09:
Hi @peitschie ,

I had a problem with the Accu Chek Mobile, now it works again.

When I tried to connect to iphone, I then failed to connect to android.

You have to disconnect the Accu Chek, then press the round button until the led goes out and it's good.

Now I need to understand why it can't find the Accu Chek under iOS.

closed 2023-09-09 by BenLaKnet

peitschie, 2023-09-10:
@BenLaKnet sounds like a lot of hard debugging work there. Congratulations on getting to the bottom of things 🙂 

For iOS... does the device bond at all? (e.g., show's up a pairing dialog)?

BenLaKnet, 2023-09-11:
It is running with iOS, I had a part of code to detect if location is available and here... I had an issue.
Not in BLE plugin.

Many thanks @peitschie