don/cordova-plugin-ble-central #990: Document interface BLEScanOptions

GerardoPrototype, 2023-09-28:
First of all, thank you very much for the work you do with this plugins <3.

I just wanted to mention if the options for this method work "startScanWithOptions".

I saw that the BLEScanOptions interface was missing some fields.

peitschie, 2023-09-28:
@GerardoPrototype if you can paste the updated docs or types you think are missing here, I'm happy to add them in! 🙂

GerardoPrototype, 2023-09-28:
I think I was wrong, the typos are in the @awesome-cordova-plugins/ble/ngx 
but I think these are the ones that should be according to the documentation

interface BLEScanOptions {
    reportDuplicates?: boolean;
    duration?: number;
    scanMode?: 'lowPower' | 'balanced' | 'lowLatency' | 'opportunistic';
    callbackType?: 'all' | 'first' | 'lost';
    matchMode?: 'aggressive' | 'sticky';
    numOfMatches?: 'one' | 'few' | 'max'
    phy?: '1m' | 'coded' | 'all'
    legacy?: boolean;
    forceScanFilter?: boolean;
    reportDelay?: number;
}

peitschie, 2023-09-28:
Right! Yeah... I am not a maintainer on that wrapper, so you are best to raise the issue on their github site over at https://github.com/danielsogl/awesome-cordova-plugins

What you've documented here is correct based of the type definitions this repo has though.

closed 2023-09-28 by peitschie