don/cordova-plugin-ble-central #992: App crashing while giving authorise and post notification permission accept - android 12 & 13, SDK 33

sivakavi-manurevasolutions, 2023-10-17:
I am facing app crash while user give authorise and post notification permission accept.

compileSdkVersion = 33
targetSdkVersion = 33

Platforms - Android 12 & 13

sivakavi-manurevasolutions, 2023-10-18:
Thanks for the Reply @peitschie.
cordova : 12.0.0
Installed platforms:
  android 10.1.2
  ios 6.3.0
  
  Will share the stack traces.

sivakavi-manurevasolutions, 2023-10-18:
Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1, result=-1, data=Intent { act=[android.content.pm](http://android.content.pm/).action.REQUEST_PERMISSIONS (has extras) }} to activity {tts.mobility.phalsbourg/tts.mobility.phalsbourg.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.apache.cordova.CallbackContext.sendPluginResult(org.apache.cordova.PluginResult)' on a null object reference

peitschie, 2023-10-18:
@sivakavi-manurevasolutions what version of this plugin are you using? This looks similar to something like #698

sivakavi-manurevasolutions, 2023-10-19:
@peitschie  plugin version : v1.3.0

peitschie, 2023-10-19:
That's a really old version of the plugin. Could you try upgrading to latest and seeing whether issue goes away? There's been quite a few fixes around this area in various release. v1.7.1 is the current version, and should work without any changes in your latest app.

sivakavi-manurevasolutions, 2023-10-19:
@peitschie  Sure, Thanks

peitschie, 2023-11-03:
I'll close this out, as the old version of the plugin (1.3.0) was unlikely to be fully compatible with the recent Android versions described here. Feel free to re-open if the current release still shows the same issue.

closed 2023-11-03 by peitschie