don/cordova-plugin-ble-central #996: Need to show Authorise popup manually

sivakavi-manurevasolutions, 2023-11-13:
I need to show the authorise popup manually and ask permission to user in app home screen. I have attached the authorise pop.
[image]

peitschie, 2023-11-13:
Hi @sivakavi-manurevasolutions 

This pop-up will automatically show when you attempt a scan. The plugin will request any runtime permissions necessary to achieve the scanning. For Android 12, you'll need to be running at least version 1.6 of this plugin. I'd highly recommend using the latest version however.

If this doesn't function for you, please feel free to re-open this and we can discuss more!

closed 2023-11-13 by peitschie

sivakavi-manurevasolutions, 2023-11-14:
@peitschie  Thanks for your reply. Yes i have used latest version 1.7.1 only. Client asked us to app should ask this permission pop in the home page itself. Is there any option to show manually this permission popup.

peitschie, 2023-11-14:
@sivakavi-manurevasolutions the best approach is using https://github.com/dpa99c/cordova-diagnostic-plugin#requestbluetoothauthorization

This plugin itself doesn't expose the necessary feature on it's own at the current state. Patches are welcome to address this however.