don/cordova-plugin-ble-central #997: Update Required for writeCharacteristic in Peripheral.java for Android 14 API 34 Compatibility

MDB80, 2023-11-15:
Dear Don,
I recently encountered an issue with the Cordova BLE Central library, specifically within the Peripheral.java file. After updating to Android 14 (API 34), I noticed that the writeCharacteristic function has been deprecated. The Android documentation provides an alternative method for this functionality.

To ensure compatibility with my app, I took the liberty of implementing the necessary changes. However, I believe it would be beneficial for the broader community if these updates could be integrated into the official library. Below, I've outlined the solution I used to address this deprecation issue.

 private void writeCharacteristic(CallbackContext callbackContext, UUID serviceUUID, UUID characteristicUUID, byte[] data, int writeType) {

        if (gatt == null) {
            callbackContext.error("BluetoothGatt is null");
            commandCompleted();
            return;
        }

        BluetoothGattService service = gatt.getService(serviceUUID);

        if (service == null) {
            callbackContext.error("Service " + serviceUUID + " not found.");
            commandCompleted();
            return;
        }

        BluetoothGattCharacteristic characteristic = findWritableCharacteristic(service, characteristicUUID, writeType);

        if (characteristic == null) {
            callbackContext.error("Characteristic " + characteristicUUID + " not found.");
            commandCompleted();
            return;
        }

        boolean success = false;

        **//characteristic.setValue(data);
        //characteristic.setWriteType(writeType);**
        synchronized(this) {
            writeCallback = callbackContext;

            **//if (gatt.writeCharacteristic(characteristic)) {  deprecated in API level 33.
            if (gatt.writeCharacteristic(characteristic,data,writeType) == BluetoothStatusCodes.SUCCESS) {**
                success = true;
            } else {
                writeCallback = null;
                callbackContext.error("Write failed");
            }
        }

        if (!success) {
            commandCompleted();
        }

    }

I hope this update can be reviewed and potentially incorporated into the library to help others facing similar compatibility challenges with Android 14.

Thank you for your consideration.
MDB80

peitschie, 2023-11-16:
Hi @MDB80, thanks for sharing the snippet!

The idea here is right, but we still want to support older versions, so the plugin would need to use proper version branching so we don't break things that don't have this API still. But... good to see an attempt 🙂

MDB80, 2023-11-17:
Hi @peitschie , thanks for your reply!
I don't mean to be impertinent, but wouldn't versioning be sufficient?
`
if (Build.VERSION.SDK_INT > 33 ) {
    // New Code 
} else {
    // Old Code
}
`
And then there's the issue that all new apps built for < API 33 will be removed, while existing apps (i.e., those already published) must target at least API 31, otherwise they will be removed from the store.

[https://support.google.com/googleplay/android-developer/answer/11926878?hl=en](url)

I don't see the point of keeping code in development that then cannot be distributed in the Play Store, especially since new apps and updates now need to target at least API 33, where my modification is required. It makes little sense to focus on backward compatibility when it limits distribution on the current Play Store.

Regardless, I have already implemented this trivial solution for my project and wanted to share it with you, thinking it might be helpful.  Thank you very much for your attention to this matter and for the work you continue to do on this plugin. Goodbye and good luck with your ongoing work.

closed 2023-11-17 by MDB80

peitschie, 2023-11-17:
> I don't see the point of keeping code in development that then cannot be distributed in the Play Store

Yep. This definitely makes sense for your use case. The challenge with Android is that due to side-loading and enterprise management tools, the requirement is less important than it seems. I know of a few applications using this library that are still only targeting APIv30 🙂.

But .. I do want to stress again, I really appreciate the report and code snippet. This work is on my radar, and contributions looks yours help bring the work along. Please reach out if you hit any other issues!