don/cordova-plugin-ble-central #999: Problem with bluno

eccessivo75, 2023-12-05:
I am facing a strange behavior on ANDROID (13):
If i scan for all devices using scan with option, i see many devices(example my esp32 project based on NimBLE) but not my BLUNO beetle.
If use a third party app  the devices are all visible also Bluno  beetle! I can connect and read all the advertisement data and the names.
The same if i ask android to scan: i see all the devices!
What am i doing wrong?

peitschie, 2023-12-05:
Hi @eccessivo75 

What is the scan command you're using? Are you seeing any errors logged in adb?

eccessivo75, 2023-12-05:
Hi and thank you for your answer:
command is :   ble.startScan([], onScanSuccess, onScanFailure);
No errors in the console log.
This is part of the code:
`app.startScan = function()
{
	app.disconnect();

	console.log('Scanning started...');

	app.devices = {};

	var htmlString =		
	'<img src="img/loader2.gif"width="30" height="30" ' +
	'style="display:inline; vertical-align:middle">' +
'<p style="display:inline;color:white">   Scanning...</p>';

	$('#scanResultView').append($(htmlString));

	$('#scanResultView').show();

	function onScanSuccess(device)
	{
        console.log(JSON.stringify(device));
		if (device.name != null)
		{
			app.devices[device.address] = device;

			console.log(
				'Found: ' + device.name + ', ' +
				device.address + ', ' + device.rssi);

			var htmlString = 			
				'<div class="deviceContainer"><ons-button modifier="large outline" onclick="app.connectTo(\'' +
				device.address + '\')">'+device.name+'</ons-button></div>'

			$('#scanResultView').append($(htmlString));
		}
	}

	function onScanFailure(errorCode)
	{
		// Show an error message to the user
		app.disconnect('Failed to scan for devices.');

		// Write debug information to console.
		console.log('Error ' + errorCode);
	}

	
 ble.startScan([], onScanSuccess, onScanFailure);
	$('#startView').hide();
};`

eccessivo75, 2023-12-05:
I can add a weird behavior: if i use the app "ble serial" on my phone (iphone12) , i can see and clone Bluno.
If i start advertising from the clone , that clone is visible from ble central.. while the original still no

peitschie, 2023-12-05:
You've got a filter there to only report devices that have a name... are you certain that the Bluno device is named? Do you get any console logs of the stringified device there?

eccessivo75, 2023-12-05:
You are coorect about names .
For debug purpose i am stringify the device data before any parsing on the  names.
What i get is a list of a variable number of stations (4 to 7 ), The most without names but only the id . The most are not connectable devices. i try take a screen shot.

peitschie, 2023-12-05:
Does the Bluno device has a specific service UUID you can use to scan for just that kind of device?

eccessivo75, 2023-12-05:
i scan for a specific local name (HEFMXV1.1). Also i am able to connect the clone. 
[image]

peitschie, 2023-12-05:
@eccessivo75 I wonder if this applies to what you're seeing here: https://github.com/don/cordova-plugin-ble-central/issues/979#issuecomment-1593706662

eccessivo75, 2023-12-05:
I try it now it will take some minutes because i am using cordova with monaca cloud service. What i miss is how to exclude this permission android:usesPermissionFlags="neverForLocation" since i can't edit android manifest from my config.xml or i get compilation errors.
It was my normal policy in the past but after the upgrade to cordova CLI 12  it is simply not possible.
Anyway when i start scanning i get a pop up asking for the run time permissions ...

peitschie, 2023-12-05:
For your first test, I'd just open the cordova app in Android studio and modify it directly to see what happens 🙂

eccessivo75, 2023-12-06:
Thank you but i have to improve my confidence with android studio... 
About the problem it seems really to be what you mentioned but i am still having the same problem.
Opening the ApK in android studio i can see that in thefile manifest.xml there is no mention of neverForLocation.
I think that the problem could be that i am not using the @slim version of the lib

eccessivo75, 2023-12-06:
I really feel like stupid: 
If i type the suggested command cordova plugin add cordova-plugin-ble-central@slim i get this error: 
Cannot find plugin.xml for plugin "cordova-plugin-ble-central". Please try adding it again.
I can install other third party plugins

peitschie, 2023-12-06:
@eccessivo75 have you tried removing the plugin first? I'm not getting this problem when I try the same on a clean project?

eccessivo75, 2023-12-06:
Thank you, yes i first removed it and then installed. i copy below the text from my command prompt :
C:\hefmx113>cordova plugin list
cordova-custom-config 5.1.1 "cordova-custom-config"
cordova-plugin-ble-central 1.7.2 "BLE"
cordova.plugins.diagnostic 7.1.4 "Diagnostic"

C:\hefmx113>cordova plugin remove cordova-plugin-ble-central
Uninstalling cordova-plugin-ble-central from android
Subproject Path: CordovaLib
Subproject Path: app
Removing "cordova-plugin-ble-central"
Removing cordova-plugin-ble-central from package.json

eccessivo75, 2023-12-06:
i can confirm the same here. I will investigate further

eccessivo75, 2023-12-07:
Actually i am not able to get rid of this. So frustrating.
I was able to use the slim version of the plugin and follow all the steps to solve as mentioned in #979.
i edit the xml, compilation goes fine, at run time i am requested with a pop up to approve the runtime permission and i consent.
but, if i include  in the xml permission neverforlocation i can see many BT station but not Bluno . As mentioned in #979 i omitted neverforlocation but ,this time scan result are completely empty! In both cases cordova diagnostic shows CONNECT and SCAN position as granted. Ideas?

peitschie, 2023-12-07:
@eccessivo75 nice detective work! It's still strange that all this was needed. What phone model and Android version was this on, out of interest?

eccessivo75, 2023-12-08:
@peitschie  thank you! it's a galaxy tab s7 fe model sm-t733 
Android 13 updated on 29 nov
For me has been more painful to work it out since i am using Monaca cloud service. 
@peitschie pointed me in the right direction : #979 was the reason why i was not able to see bluno.
So for the one that are struggling as i was here is (not maybe the perfect) solution:
I installed the  cordova-plugin-ble-central@slim. Monaca is not able to install the slim version by its own, so you have to download it (maybe there is a direct link... i used instead cordova CLI on my computer), created a zip archive, then uploaded the archive as a custom plugin in monaca plugin manager.
in config.xml add:
<config-file target="AndroidManifest.xml" parent="/manifest">
        <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" />
        <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
        <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
        <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />
        <uses-permission android:name="android.permission.BLUETOOTH_SCAN"  />
        <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    </config-file>
 Then you need to ask  the runtime permission for the fine location. I used, cordova diagnostic plugin and inside my "app.js" file in the device ready event handler at the very first place i added:

cordova.plugins.diagnostic.requestRuntimePermission(function(status){
    switch(status){
        case cordova.plugins.diagnostic.permissionStatus.GRANTED:
            console.log("Permission granted to use the bt");
            break;
        case cordova.plugins.diagnostic.permissionStatus.NOT_REQUESTED:
            console.log("Permission to use the bt has not been requested yet");
            break;
        case cordova.plugins.diagnostic.permissionStatus.DENIED_ONCe:
            console.log("Permission denied to use the bt - ask again?");
            break;
        case cordova.plugins.diagnostic.permissionStatus.DENIED_ALWAYS:
            console.log("Permission permanently denied to use the bt!");
            break;
    }
}, function(error){
    console.error("The following error occurred: "+error);
}, cordova.plugins.diagnostic.permission.ACCESS_FINE_LOCATION);

closed 2024-03-12 by peitschie