don/cordova-plugin-ble-central #1000: The plugin ble.write in versions after iOS 17 reported an error

zhanLaiDong, 2023-12-29:
Hello author, first of all, thank you for your contribution. After the iOS version was updated to 17, an error occurred in ble.write when Bluetooth was connected to the printer for printing. May I ask how to solve this problem? I cannot know exactly what information the user has lost, and I cannot reproduce it. I am very anxious. Please, thank you very much
[image]

Could not find service with UUID FFF0 on peripheral with UUID AC9E3C01-1EC0-1172-6E27-36A516479242

zhanLaiDong, 2024-01-07:
Please, brothers, wish you all a safe life

peitschie, 2024-01-08:
Hi @zhanLaiDong 

Can you paste what your javascript code wiring all this together looks like? In the above console screenshot there, it says the state is `connecting` still.

> The device's status kept showing that it was connected

The phone may be connected but the app is not aware of this. Standard practice is to always call `ble.connect` on app launch and after any kind of disconnect.

zhanLaiDong, 2024-01-08:
Hello, the current issue is that the customer did not continue printing for a short period of time after using Bluetooth to connect to the printer. For example, after about 20 minutes, this problem may occur again when printing. This is a random occurrence. The more printers and mobile phones there are around the customer, the higher the probability of errors. Thank you
 

幸运信息
***@***.***

zhanLaiDong, 2024-01-08:
Also, restarting the printer has no effect. Only after the app kills the process and reopens it can it print normally

幸运信息
***@***.***

zhanLaiDong, 2024-01-08:
After each printing of the app, it needs to be disconnected because there are multiple phones sharing the same printer device, so it cannot be kept connected all the time, which means other phones cannot connect to the printer

peitschie, 2024-01-08:
@zhanLaiDong without examples of how the javascript connection loop there is working, it's fairly difficult to guess what's going on.

zhanLaiDong, 2024-01-09:
This is a normal printed code, but there may be occasional printing errors. Please check, thank you
[code.docx](https://github.com/don/cordova-plugin-ble-central/files/13870750/code.docx)

peitschie, 2024-01-10:
So, a few things:

1. The service can fail to be found if the printer is no longer connected by the time you attempt to write. This might be what's happening here, as you can see the `CBPeripheral` state is `connecting` in your screenshots above.
2. I notice your `doPrint` method is allowing both a `Write` and `WriteWithoutResponse` method to be the `writeService`, however, you only ever call `ble.write`. This means that if you find `WriteWithoutResponse` for some reason, the write call will fail because you should be using `ble.writeWithoutResponse`

As an aside, the whole approach there to finding the characteristic is very unusual. Usually, the service and characteristic UUIDs are a fixed part of the contract, and don't need to be extracted like you are doing here.

peitschie, 2024-03-12:
Closing this out due to inactivity. Feel free to re-open if there is still an unresolved issue here.

closed 2024-03-12 by peitschie