don/cordova-plugin-ble-central #1004: Include bluetooth serial plugin in this repo

lajith111, 2024-02-26:
This plugin only supports BLE device..Now  Most of the thermal printers works with serial plugin.. The other plugin not up to date.

@peitschie Is it possible to include serial connection  (some featues like conect /read/write ) etc..it will be more helpful

https://github.com/don/BluetoothSerial

Thanks in advance..

peitschie, 2024-02-27:
Hi @lajith111 

There's a few extra dependencies and the like related to Bluetooth serial support that wouldn't be desirable to spread to all users of this plugin. I'm afraid there's little I can do to help there, as I don't have any capacity to maintain the additional plugin.

closed 2024-02-27 by peitschie

lajith111, 2024-02-28:
Thanks for the reply.. the other serial plugin does not have a maintainer.. I have observed some are ready to maintain some permission changes related to android update.. Mr @don  is not replaying for any query . Can you help on this

this will be great help from your side