don/cordova-plugin-ble-central #1013: Does this plug-in support Bluetooth 4.0 HM10 module?

bluemoon-i, 2024-05-30:
Does this plug-in support Bluetooth 4.0 HM10 module?
After connecting to Bluetooth, failure when using ble.startNotification

peitschie, 2024-05-30:
Hi @bluemoon-i 

Presuming this is a Bluetooth LE module, there should be no problems with support.

Are you using this on iOS or Android (or both)?

If Android, I'd highly recommend trying to connect and interact using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) as a nice way to double-check that the notification information is properly available for the BLE peripheral.

bluemoon-i, 2024-05-31:
[image]

bluemoon-i, 2024-05-31:
I'm using Android, and it seems like I can receive notifications normally, but when I write '1', I get an error

peitschie, 2024-05-31:
That looks like the peripheral itself has some issues. 

Given the response there is coming from the Android layer itself, it's not going to be caused by anything in this plugin.

I can't really help much with this I'm afraid. You'd need to contact whoever made the BLE device for support here.

bluemoon-i, 2024-05-31:
OK, thanks for your answer

closed 2024-05-31 by peitschie