don/cordova-plugin-ble-central #1014: NSInvalidArgumentException thrown in findPeripheralByUUID

kenguest, 2024-06-04:
Hi, I've gotten a few reports via Sentry.io of NSInvalidArgumentException being thrown in findPeripheralByUUID under iOS.

Exception message is: "[__NSCFBoolean isEqualToString:]: unrecognized selector sent to instance 0x20e8e3ce8"

- (CBPeripheral*)findPeripheralByUUID:(NSString*)uuid {
    CBPeripheral *peripheral = nil;

    for (CBPeripheral *p in peripherals) {

        NSString* other = p.identifier.UUIDString;

        if ([uuid isEqualToString:other]) {
            peripheral = p;
            break;
        }
    }
    return peripheral;
}

Stacktraces are showing this occurring when findPeripheralByUUID  is called from isConnected.

I'm unfamiliar with Objective C so I'm not in a position to provide a patch for fixing this.

peitschie, 2024-06-05:
Thanks @kenguest 

Do you have an example of what the string your searching for is (i.e,. the UUID you're supplying to the code there)?

I agree the plugin shouldn't be crashing for this!

kenguest, 2024-06-13:
we're passing the bluetooth mac address, one such uuid is 12:A1:8B:EC:86:CC

peitschie, 2024-06-17:
@kenguest that looks to me like a MAC address rather than a uuid.
A uuid is usually of a form more like: `b111d2f9-2cb5-4541-9119-015b8836d0fb`

On Android, the ID reported by the phone is often the MAC address found in the broadcast packet. On iOS however, this is normally a guid, and is not predictable between installs.

Are you locating your device via a `scan` before trying to connect? Or are you wanting to connect to a device with a known MAC address on iOS?

peitschie, 2024-07-10:
Hrm... this is definitely a mystery, as iOS has no way to map the MAC address back to the UUID it demands for connection.

I'll have a look tomorrow at making this plugin more clearly error out rather than crashing when handed a non-UUID identifier.

But... I don't see how this fixes your case here specifically, as I don't really understand how your approach of calling `ble.connect(macAddress)` ever worked on iOS 🙂

closed 2024-07-14 by peitschie

peitschie, 2024-07-14:
I've pushed a fix for this in cordova-plugin-ble-central@1.7.5-alpha.0

@kenguest are you able to see if this addresses the issue you're seeing here?

kenguest, 2024-07-17:
thanks - I'll see if it does.
To revert back to your question, you're absolutely correct - ble.connect() under iOS is taking values such as  this guid 9DA4BDEB-3AB1-2066-8C99-0F8991036CED  to denote what device to connect to whereas Android uses mac addresses.

PR #1021 (2024-07-14): Assorted iOS fixes
Closes #1014, #905, #919, #741