don/cordova-plugin-ble-central #1025: Error when importing cordova-plugin-blöe-central

drolpi, 2024-07-31:
Hello, as soon as I use this library I get this error, what could be the reason? Please help! 

> [vite] 13:00:02 [vite] Internal server error: Failed to resolve entry for package "cordova-plugin-ble-central". The package may have incorrect main/module/exports specified in its package.json.
[vite]   Plugin: vite:import-analysis
[vite]   File: C:/Users/larsn/WebstormProjects/GoMoof/src/lib/ble/bikeConnector.ts
[vite]       at packageEntryFailure (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:49442:17)
[vite]       at resolvePackageEntry (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:49439:5)
[vite]       at tryNodeResolve (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:49209:20)
[vite]       at Context.resolveId (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:48959:28)
[vite]       at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
[vite]       at async Object.resolveId (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:52242:32)
[vite]       at async TransformContext.resolve (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:51933:23)
[vite]       at async normalizeUrl (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:67297:34)
[vite]       at async file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:67461:47
[vite]       at async Promise.all (index 0)
[vite]       at async TransformContext.transform (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:67382:13)
[vite]       at async Object.transform (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:52318:30)
[vite]       at async loadAndTransform (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:55073:29)
[vite]       at async viteTransformMiddleware (file:///C:/Users/larsn/WebstormProjects/GoMoof/node_modules/vite/dist/node/chunks/dep-DEPSZ3SS.js:64920:32)

Sammy051, 2024-07-31:
**We're using:**
Capacitor: 6.0.0
Plugin: cordova-plugin-ble-central@1.7.4
**Fails in:**
`ionic serve`
`ionic cap run android -l --external`
`ionic cap run android`
`ionic cap run ios -l --external`

peitschie, 2024-07-31:
Unlike a native Capacitor plugin, for cordova you shouldn't have any import references to the plugin anywhere in code. The plugin js should be integrating via Cordova's build mechanism... I wouldn't expect to see vite here.

Do you have any import statements anywhere?