don/cordova-plugin-ble-central #1026: startNotification results in  Could not find service with UUID 0001

markarupert, 2024-08-12:
I am getting this error on my iOS app. I have been using the plugin for years and this suddenly came up. So I'm not sure what changed. 

"Could not find service with UUID 0001 on peripheral with UUID XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX"

I am using version 1.7.4.

Here is the start notification code.

```

service = "0001";
characteristic = "0001";
ble.startNotification(mmaddr,
    service,
    characteristic,
    ot.ioSixNotifyFcn,
    function (e) {
        deviceLog('BLE: startIOSixBLE: Failed to startnotification: ' + mydump(e), function (str) {console.log(str)});
    });
```

peitschie, 2024-08-13:
Hi @markarupert 

Is the problem intermittent, or consistent across all devices? Also, what version were you on previously? 

There haven't been many changes to the iOS code for quite some time, but I have found sometimes that iOS updates cause issues like this.

The other thing to try is whether 1.7.5-alpha.0 makes any impact here. I [recently added additional logging](https://github.com/don/cordova-plugin-ble-central/commit/28a610382a4fa2200bd71ff14e224fd6cc236fca) to the alpha package around service discovery failures, which perhaps could point to more insight as to what's happening here.

markarupert, 2024-08-13:
Thanks for the response.

It is happening on both my devices. It seems like it is a project issue. It previously worked with 1.4.4 but now it isn't. I updated to the latest version and got the same result. It is happening on all my test devices.

I tried updating with:

cordova plugin add cordova-plugin-ble-central@1.7.5-alpha.0 

It is failing to fetch the plugin.

peitschie, 2024-08-14:
> It previously worked with 1.4.4 but now it isn't

Just to double-check my understanding, are you saying that your application no longer works with version 1.4.4 either?

markarupert, 2024-08-16:
Sorry I have been doing some testing.

Yes it does not work with either version. 

I have been doing some analysis. I have several apps that use ble central. They are very similar just adjusted from certain markets. Some are working fine.

I installed the alpha version above and here is what looks like the extra logging information.
The Service and characteristic being passed to startNotification is "0001" for both

Does this help:

2024-08-16 14:46:38.504458-0400 GeoWizBiz[1113:343044] registering for notification
2024-08-16 14:46:38.504509-0400 GeoWizBiz[1113:343044] getData
2024-08-16 14:46:38.504544-0400 GeoWizBiz[1113:343044] getUUID
2024-08-16 14:46:38.504588-0400 GeoWizBiz[1113:343044] Malformed UUID:
2024-08-16 14:46:38.506618-0400 GeoWizBiz[1113:343044] readRSSI
2024-08-16 14:46:38.506876-0400 GeoWizBiz[1113:343044] getUUID
2024-08-16 14:46:38.507041-0400 GeoWizBiz[1113:343044] Malformed UUID:

peitschie, 2024-08-19:
Hi @markarupert 

The fact this has stopped working with the previous version too makes me suspect either something iOS version related... or potentially iOS has cached some bad service characteristics.

Resetting the iOS BLE cache is a bit painful though, as it requires clearing all data off the device (see https://stackoverflow.com/a/47129088).

I have found this causing issues before, especially when I'm actively developing a peripheral and changing the services/characteristics. Sometimes iOS caches an incorrect discovery payload, and can't be easily convinced that it's no longer applicable (without adding support to the peripheral for the service changed characteristic).

markarupert, 2024-08-19:
Wow, that's drastic and obviously impractical. There must be a better way.

peitschie, 2024-08-19:
@markarupert of course this is impractical as an end-user instruction. The main point of interest here is whether this has any impact on your local situation.

The fact that a previously functional version (1.4.4) is no longer working highly suggests something outside of this plugin is at fault. Until you can figure this out... it's pretty hard to guess what the correct recovery steps are.

> 2024-08-16 14:46:38.506618-0400 GeoWizBiz[1113:343044] readRSSI
> 2024-08-16 14:46:38.506876-0400 GeoWizBiz[1113:343044] getUUID
> 2024-08-16 14:46:38.507041-0400 GeoWizBiz[1113:343044] Malformed UUID:

It's interesting that you're seeing a Malformed UUID here, as the only ID supplied to `readRSSI` is a device identifier.
Are you certain the scan is picking up a valid device?

markarupert, 2024-08-20:
It does seem to be having trouble with the scan. I got it to work by backing down the version of cordova-ios to 5.1.1.

peitschie, 2024-08-21:
Is there any chance the phone or some other nearby device is sometimes connecting to the device you're scanning for?

There's been no real changes within the plugin that would impact the scanning behaviours for a long time, so I can't really guess why jumping back to cordova-ios 5.1.1 (which is several years old) would have any real impact here.

Certainly a mystery!

peitschie, 2024-08-28:
Makes sense. I'll leave the ticket open for a while then... we'll see if anything else interesting crops up in your research when you get time 🙂

peitschie, 2024-10-26:
Hi @markarupert 

I wonder if the issue you see here might be connected to #1039.

I've made an update in 1.7.9 (currently alpha) that may improve this situation if it is. If you have time, you can try out the fix by `npm install cordova-plugin-ble-central@1.7.9-alpha.1`

peitschie, 2025-02-13:
I suspect this is the same underlying condition as https://github.com/don/cordova-plugin-ble-central/issues/1039
Marking as resolved as a fix is released in 1.7.9.

Please re-open if the problem still persists.

closed 2025-02-13 by peitschie