don/cordova-plugin-ble-central #1027: Failed to subscribe to notification (sometimes)

graphefruit, 2024-08-20:
Hello together, @peitschie  :)
(first of all, sorry for taking that long to open up the ticket)

I wanted to say that I'm deeply thankful for your support here and making this awesome plugin, maintaining it so wonderfull and helping us all out! THANKS!!!!

To the situation:
Somehow I got a bluetooth scale which is doing strange things on customers side.
Most of the times, the app can connect to the bluetooth scale.
But sometimes, the app can't connect to the bluetooth scale.
The issue comes from the ble.startNotification

 ```"DEFAULT: [\"failed to subscribe to notifications \\\"Could not find service with UUID 49535343-fe7d-4ae5-8fa9-9fafd205e455 on peripheral with UUID 140FC66D-B212-05F1-F5C1-2A285331FC85\\\"\"]",```

If the user restarts the app, then the app can connect to the given UUID and Notification ID.
Also to mention that when trying to `ble.connect` again after some seconds and try again to `ble.startNotification` don't resolve the issue, you realy need to restart the app.

I've also tried to add a timeout with 150ms before calling ble.startNotification but this doesn't help either.

Adding to this, when the connection fails, I set a timeout, I search for the device again, and just when its found I try to connect and give also there needed timeouts.

Do you have any ideas why this behaves that strange?

Furthermore heres a screenshot of two logfiles: The left side where the connection works without any issues, the right one where the notification fails:
[image]

Thanks for your great work and your awesome support!
Lars

peitschie, 2024-08-21:
Hi @graphefruit 

Which version of the plugin are you running with this? Is there any chance you can test a build with the current alpha version (cordova-plugin-ble-central@1.7.5-alpha.1)?

There's been a commit recently that might make it more obvious if there's a service discovery failure going on here: https://github.com/don/cordova-plugin-ble-central/commit/28a610382a4fa2200bd71ff14e224fd6cc236fca

graphefruit, 2024-08-21:
Hey @peitschie, 
thanks for the fast support and response!
Actually I'm running on `cordova-plugin-ble-central 1.7.4 "BLE"`
I will install this plugin and will sent it to the tester/person which can reproduce this issue sometimes.

I guess the error is passed to cordova, so I can rightly log it correct?

graphefruit, 2024-08-26:
@peitschie sad news :(
I've installed the cordova-plugin-ble-central@1.7.5-alpha.1, with one scale it works to transmit the weight to the app.
The other scale is not working anymore. Reverting back to 1.7.4 let the scale work again.

Can I do something to send you debugging helps? (Shall I create a new ticket aswell?)

peitschie, 2024-08-28:
Hi @graphefruit 

Grabbing debug logs on iOS is very difficult, unfortunately. The fact this is intermittent with the same version of the plugin does point towards something about the peripheral causing issues with iOS, rather than the plugin itself here having any issue.

It's worth checking if another BLE app has the same issue (e.g., [nrf Connect](https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403)) as this would hint much more strongly at a peripheral issue.

graphefruit, 2024-08-28:
Good Morning @peitschie !
Thanks for the feedback.

Maybe I wrote a bit miss understanding, sorry for this, I'll try to clear it:

Scale 1:
I have a scale: DiFluid Microbalance TI.

On 1.7.4 it connects and the app can retrieve the send weight from the scale.
On 1.7.5 it connects but the app **CANT** retrieve the send weight anymore.

Scale 2:
I have a scale: Acaia Pearl S

On 1.7.4 it connects and the app can retrieve the send weight from the scale.
On 1.7.5 it connects and the app can retrieve the send weight.

The Acaia Pearl S was the scale which had issues that sometimes the App can't connect to the notifications and needed a reopening of the app completly.

Cause of Scale 1: It somehow points out the fact that something was changed in the version, which now makes somehow issues with attaching to the notification and get all data to handle.

Thanks for your help!
Have a great cup of morning coffee
Lars

graphefruit, 2024-10-06:
Hello there, coming back to this.
I've installed Version 1.7.8 and both scales are sending weights again :)

I'll build soon a new version now for the user to test and coming hopefully back with the conclusion: Everything works ;) 
Thanks!
Lars

peitschie, 2024-10-06:
That's interesting to hear. It sounds like this was connected to the recent regression with UUID handling. But... the error messages you were reporting here aren't what I would have expected.

I'll close this out as it sounds like the https://github.com/don/cordova-plugin-ble-central/releases/tag/v1.7.8 has addressed this. Feel free to re-open if the problem still occurs!

closed 2024-10-06 by peitschie