don/cordova-plugin-ble-central #1029: some users experiencing "cordova is not defined"

sithwarrior, 2024-08-30:
We are experiencing some users that hit the following error (small percentage)

cordova is not defined

I suspect that it might be caused by this plugin calling cordova directly, the same issue seems to have happened in another plugin outlined here: https://github.com/EddyVerbruggen/SocialSharing-PhoneGap-Plugin/issues/243

in short its caused by the plugin calling `cordova.exec` directly instead of first `const exec = require('cordova/exec');`

As also mentioned in the issue, the official plugins use this pattern

peitschie, 2024-08-31:
@sithwarrior I'd be happy to accept a PR for this if you have time?

sithwarrior, 2024-09-02:
@peitschie  Made a PR and tested the change in my own app without issue

closed 2024-09-08 by peitschie

peitschie, 2024-09-08:
This is released in 1.7.6

Sorry for the delay!

PR #1030 (2024-09-08): Update ble.js to use require format
This changes the plugin so instead of calling cordova directly, it requires cordova/exec and uses it for calling.

This also mimics how its done in plugins from the core cordova team.

fixes #1029