don/cordova-plugin-ble-central #1031: Has 1.7.5 broken 16 bit service UUID's on iOS?

ml-imas, 2024-09-09:
Hi,

The fixes in #1021 for #905 and #1014 seem to have broken 16 bit service UUID's like 180a (Device Information Service). Calling `ble.read` with "180a" as service_uuid used to work fine, but now results in an error saying

Malformed UUID: 180a

This looks like l. 1104 in `BLECentralPlugin.m` , and it matches with the change from using CBUUID's `UUIDWithString` to NSUUID's `initWithUUIDString` .

Using the full 128 bit UUID (like in `UUIDHelper.java`) doesn't work. The UUID is no longer rejected but we end up with

Could not find service with UUID 0000180A-[...]

instead; it doesn't seem to match the services reported by the device.

gwhenne, 2024-09-09:
We're having the same issue with 1.7.5. Here is what is showing on the Safari remote console:

[image]

sam-higgs, 2024-09-09:
Also experiencing this issue with versions 1.7.5 and 1.7.6.

peitschie, 2024-09-09:
Can confirm the regression. Will get out a fix shortly.

peitschie, 2024-09-10:
A fix has been released in 1.7.7 (and I've added some more test cases to cover this!)

Thanks for the quick reports!

closed 2024-09-10 by peitschie

peitschie, 2024-09-10:
@gwhenne is this issue on iOS or Android?
Are you able to raise a new issue with some more detail, and potentially confirm if the behaviour was different with a previous version of the plugin?

What you are describing sounds unrelated to this particular regression.

gwhenne, 2024-09-12:
Thank you. Yes, it does appear to be a different problem. A new issue has been opened here:

https://github.com/don/cordova-plugin-ble-central/issues/1035