don/cordova-plugin-ble-central #1033: isConnected always returns false

markamccorkle, 2024-09-11:
On the latest 1.7.7, calling `isConnected` on iOS is always returning false. Reverting to 1.7.4 fixes the issue.

youssefadel94, 2024-09-11:
reverting to 1.7.4 causes the issue of malformed service uuid

peitschie, 2024-09-11:
@markamccorkle is this just using `ble.isConnected`? Or are you using `ble.withPromises.isConnected`?

@youssefadel94 are you able to share the exact error you're seeing there? 1.7.4 did not have the UUID validation code in it, so I wonder what the issue you're seeing there.

markamccorkle, 2024-09-11:
I can dig in further to see what the ionic native wrapper uses, but this is the latest ionic capacitor.

peitschie, 2024-09-12:
Found the issue! MR up shortly

peitschie, 2024-09-12:
Fix is released in version 1.7.8

closed 2024-09-12 by peitschie