don/cordova-plugin-ble-central #1035: iOS: Attribute not found message

gwhenne, 2024-09-12:
`ble.startNotification` shows the message "The attribute could not be found" in the Console.

Although it looks connected and works okay, `ble.isConnected` fails. 

Once the device turns off, the app receives the "Peripheral disconnected" message.

This started happening in 1.4.4 on iOS.

peitschie, 2024-10-18:
Hi @gwhenne 

Which console are you meaning? Is this the web console, or XCode's debug console? Are you able to post any log snippets so we can see what activities are happening around this?

Thanks!

gwhenne, 2024-10-18:
It's on the Safari Remote Console. There are no other messages. 

Is it working OK for you?

peitschie, 2025-02-13:
I suspect this is the same underlying condition as #1039 
Marking as resolved as a fix is released in 1.7.9.

Please re-open if the problem still persists.