don/cordova-plugin-ble-central #1037: Support Question: Stick on Cordova Plugin or go to Capacitor Plugin?

graphefruit, 2024-10-06:
Hello again @peitschie,
I can't tell you often enough how wonderful this plugin works and is maintained at your side!
Thank you so much again!

The last days I got a contributor, which helped me to lift Beanconqueror from Cordova to Capacitor.
We lifted now many plugins from Cordova to Capacitor but actually my decision was to stick to this cordova-plugin-ble-central and to not update to https://github.com/capacitor-community/bluetooth-le right now.

You may have in mind that we discussed very long about performance, connection & reconnecting situations with my bluetooth pressure sensors and so on.

Therefore I wanted to give it a try to ask you if there are any upsides or also downsides going to the new Capacitor plugin.
Does it maybe have not some features this plugin has? Is the new plugin as fast as this (or maybe even faster?)

Actually some issues (even I didn't go far down the road) are holding me a bit of wanting to swap (disconnecting, timeouts, connection issues)

I've just updated to 1.7.8. and its working flawless on my android and iOS devices so far on a capacitor base.

Is it planned to maintain this plugin any longer on a mid-term or to fully focus on capacitor bluetooth-le?

Thank you so far for your time taking here and giving me some more thoughts about where to go ;)
Have a great cup of coffee
Lars

peitschie, 2024-10-06:
Hiya @graphefruit 

This plugin isn't going anywhere just yet! I've switched to capacitor for my own apps, but still am using this plugin. As long as capacitor makes it straightforward to use cordova plugins, I still plan to maintain this plugin for a while yet in terms of platform compatibility and bug fixes.

I've only just recently started dabbling with the capacitor bluetooth-le plugin. However, that plugin lacks the L2CAP connection that I require, and I can't yet predict the time frame for porting this across. As you have also noted, there are some threading differences between the two plugins which impacts performance to some level, but I haven't got enough real-world experience to pinpoint exactly where and how.

For the immediate question though: this plugin will continue to be maintained for a while still, and I have no plans switch across any time soon. You should feel confident continuing to use this plugin.

graphefruit, 2024-10-06:
Awesome!
Big thanks for the fast and detailed response!
Happy to stay here so :D
Have a great weekend!

closed 2024-10-18 by peitschie