don/cordova-plugin-ble-central #1039: iOS Bluetooth connection issue 'Could not find service with UUID 569A1101-B87F-490C-92CB-11BA5EA5167C on peripheral with UUID 2A4E24EC-B4BD-985A-9791-8933C28FDD6D'

dgrothman, 2024-10-16:
While troubleshooting some issues that have been reported in our app we have found the following issue intermittently occurs.

While trying to discover the service and characteristics the driver successfully reports the device, service and characteristics, but then right after reports that it can not find the service, and once we receive this message there is no way to recover the bluetooth connection and get it to work.

`ClaApp	didDiscoverServices\	
bluetoothd	Received XPC message "CBMsgIdServiceDiscoverCharacteristics" from session "com.pulsegroup.claapp-central-3354-353"\	
bluetoothd	Finding all characteristics in range [0x000c, 0x0016] on device "2A4E24EC-B4BD-985A-9791-8933C28FDD6D"\	
bluetoothd	Update devices: connected, BLE, 1 total, power sources 0\	
bluetoothd	Device found: CBDevice 39BA8B4E-9412-549A-576F-C732E3F8AE48, BDA 48:E1:5C:97:E3:FB, Nm 'Living Room', IDS 47DB586B-2670-4F86-88EA-A4AFBC3D7C23, stID 47DB586B-2670-4F86-88EA-A4AFBC3D7C23, DsFl 0x800000 < Pairing >, DvF 0xB000 < Hidden BLEPaired CloudPaired >, CF 0x00000000 < Attributes >\	
bluetoothd	Device found: CBDevice 6BD2C5FF-0464-87E9-0211-692659333094, BDA F2:8E:E4:AC:85:4A, Nm 'Andy\'92s MacBook Pro', IDS 90DCC38E-42CF-4A8E-B0BE-AFCA77174D79, stID 90DCC38E-42CF-4A8E-B0BE-AFCA77174D79, DsFl 0x800000 < Pairing >, DvF 0xB000 < Hidden BLEPaired CloudPaired >, CF 0x00000000 < Attributes >\	
bluetoothd	Device found: CBDevice 520D6B24-FC3D-B5DF-422A-2407EFCF0931, BDA 4C:AB:4F:94:85:1E, Nm 'Andy's room', IDS 50FCF25B-3419-4AD2-B0FD-0B4050710131, stID 50FCF25B-3419-4AD2-B0FD-0B4050710131, DsFl 0x800000 < Pairing >, DvF 0xB000 < Hidden BLEPaired CloudPaired >, CF 0x00000000 < Attributes >\	
bluetoothd	Device found: CBDevice 33274F79-C042-188E-85BF-D70CA63259D7, BDA 14:1B:A0:D6:64:EC, Nm 'INSiGHTBridge-ff-ff-ff-ff-ff-ff', IDS DB85B7D8-1007-46FF-8FBA-E513A48901AF, stID DB85B7D8-1007-46FF-8FBA-E513A48901AF, DsFl 0x800000 < Pairing >, DvF 0xB000 < Hidden BLEPaired CloudPaired >, CF 0x00000000 < Attributes >\	
bluetoothd	statedump: 0x0001 Primary Service      [ serviceUUID: 0x1801, endHandle: 0x0004, discoveredCharacteristics: some ]\	
bluetoothd	statedump: 0x0002 Characteristic       [ valueUUID: 0x2A05, valueHandle: 0x0003, properties: read indicate, discoveredDescriptors: all ]\	
bluetoothd	statedump: 0x0004 Client Configuration [ none ]\	
bluetoothd	statedump: 0x0005 Primary Service      [ serviceUUID: 0x1800, endHandle: 0x000B, discoveredCharacteristics: none ]\	
bluetoothd	statedump: 0x000C Primary Service      [ serviceUUID: 569A1101-B87F-490C-92CB-11BA5EA5167C, endHandle: 0x0016, discoveredCharacteristics: all ]\	
bluetoothd	statedump: 0x000D Characteristic       [ valueUUID: 569A2000-B87F-490C-92CB-11BA5EA5167C, valueHandle: 0x000E, properties: notify, discoveredDescriptors: none ]\	
bluetoothd	statedump: 0x0010 Characteristic       [ valueUUID: 569A2001-B87F-490C-92CB-11BA5EA5167C, valueHandle: 0x0011, properties: write, discoveredDescriptors: none ]\	
bluetoothd	statedump: 0x0012 Characteristic       [ valueUUID: 569A2002-B87F-490C-92CB-11BA5EA5167C, valueHandle: 0x0013, properties: notify, discoveredDescriptors: none ]\	
bluetoothd	statedump: 0x0015 Characteristic       [ valueUUID: 569A2003-B87F-490C-92CB-11BA5EA5167C, valueHandle: 0x0016, properties: write, discoveredDescriptors: none ]\	
ClaApp	didDiscoverCharacteristicsForService\	
ClaApp	Found characteristics for service <CBService: 0x00000000, isPrimary = YES, UUID = 569A1101-B87F-490C-92CB-11BA5EA5167C>\	
ClaApp	Characteristic <CBCharacteristic: 0x00000000, UUID = 569A2000-B87F-490C-92CB-11BA5EA5167C, properties = 0x10, value = (null), notifying = NO>\	
ClaApp	Characteristic <CBCharacteristic: 0x00000000, UUID = 569A2001-B87F-490C-92CB-11BA5EA5167C, properties = 0x8, value = (null), notifying = NO>\	
ClaApp	Characteristic <CBCharacteristic: 0x00000000, UUID = 569A2002-B87F-490C-92CB-11BA5EA5167C, properties = 0x10, value = (null), notifying = NO>\	
ClaApp	Characteristic <CBCharacteristic: 0x00000000, UUID = 569A2003-B87F-490C-92CB-11BA5EA5167C, properties = 0x8, value = (null), notifying = NO>\	
ClaApp	0x00000000 - [PID=3470, throttler=0x00000000] ProcessThrottler::Activity::Activity: Starting foreground activity / 'WebPageProxy::runJavaScriptInFrameInScriptWorld'\	
ClaApp	registering for notification\	
ClaApp	getData\	
ClaApp	Could not find service with UUID 569A1101-B87F-490C-92CB-11BA5EA5167C on peripheral with UUID 2A4E24EC-B4BD-985A-9791-8933C28FDD6D\	
ClaApp	0x00000000 - [PID=3470, throttler=0x00000000] ProcessThrottler::Activity::Activity: Starting foreground activity / 'WebPageProxy::runJavaScriptInFrameInScriptWorld'\	
ClaApp	getData\	
ClaApp	Could not find service with UUID 569A1101-B87F-490C-92CB-11BA5EA5167C on peripheral with UUID 2A4E24EC-B4BD-985A-9791-8933C28FDD6D\	
ClaApp	0x00000000 - [PID=3470, throttler=0x00000000] ProcessThrottler::Activity::invalidate: Ending foreground activity / 'WebPageProxy::runJavaScriptInFrameInScriptWorld'\	
ClaApp	0x00000000 - [PID=3470, throttler=0x00000000] ProcessThrottler::Activity::Activity: Starting foreground activity / 'WebPageProxy::runJavaScriptInFrameInScriptWorld'\	
ClaApp	stop notification\	
ClaApp	getData\	`

What are some ways we can troubleshoot further to find the cause, and / or what would be the cause of this seemingly failed service lookup?

peitschie, 2024-10-18:
Hi @dgrothman 

That does definitely look a little strange.

Which version of the plugin are you on? Do you see this issue with another application (e.g., [nrf Connect](https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403))?

The only bit of local debugging I could suggest is to adjust the loop at https://github.com/don/cordova-plugin-ble-central/blob/860455c0bf4af099fe44fd8c7dcfec1f924c1162/src/ios/BLECentralPlugin.m#L1016 to see if there are potentially multiple `peripheral` objects there with a matching UUID somehow?! I've never heard of this being a thing though...

dgrothman, 2024-10-18:
We are currently running 1.7.3 version of the driver. We just were able to get some good debugging data and the following image seems to validate your idea that there are multiple peripheral objects with the same UUID

[image]

This shows both peripherals one with the services which is used for the log message and one without the services which is used when trying to connect to the service. The only way we are able to reliably reproduce this is by connecting to the device and using it and then leaving it overnight, then when we connect to the device again in the morning it will be in this failed state.

As we are not objective C developers, is this something that looks like it is possible to patch?

dgrothman, 2024-10-21:
Hi @peitschie  I did just complete another test with the latest version of the driver 1.7.8 and verified the issue is still present
Here is a screen capture of the peripherals have a "ghost" version of a device with no services

[image]

peitschie, 2024-10-25:
Hi @dgrothman 

I've pushed up a test MR that I've only very mildly tested... are you able to see if this helps solve your issue?

If possible, I'd really appreciate any help regression testing the full range of impacted APIs on iOS.

dgrothman, 2024-10-25:
Thank you, I will work on this early next week and get back to you with the results.

peitschie, 2025-02-13:
@dgrothman just checking in, did you ever get a chance to test out the new patch?

peitschie, 2025-02-13:
Marking as resolved as a fix is released in 1.7.9.

Please re-open if the problem still persists.

closed 2025-02-13 by peitschie

dgrothman, 2025-02-14:
Hey @peitschie, sorry I thought I responded. We have pushed the patch and it is working great! Thank you again for your support with this.