don/cordova-plugin-ble-central #1043: connect to mi band 4-5-6

advancedcodingenterprise, 2025-01-12:
I see it in many newer BLE devices they require confirmation by the device itself for the connection. Which means I have to invoke some sort of write function (I guess) so I can establish connection.
The example I can give and I have troubles is with Mi Smart Band 4.
I am bonding the devices - connecting then when I try to subscribe for heartrate or read hearrate data (on appropriate characteristics) I just don't receive anything. I was wondering if the device required PIN or if I am missing anything since all connections are successfull except reading the data.
Then I saw that the native app of the band forces somehow the band to pop a confirmation for the connection. Now can anyone give me a clue how this can happen? May be I have to write on some characteristics?

clivinn-shla81092, 2025-01-12:
Hello, we tried to solve the issue.

This is what we did:

Implement a new function 'confirmConnection' in the BLE plugin to send a connection confirmation request to the device after bonding. This function will write to a specific characteristic to trigger the confirmation prompt on the device.

You can review changes in this commit: https://github.com/clivinn-shla81092/don-cordova-plugin-ble-central-1043/commit/51b785dc9c8484f64b5f8da4f56582054477c7cc.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

graphefruit, 2025-01-22:
So after you can establish the connection but don't receive any data:
What happens when you write some kind of empty bytes to the device after connection?

I know this from my bluetooth devices, where some devices needs a "starting" signal to send the data then after connection.
But I think you need to find out which exact command you need to write.

dariosalvi78, 2025-03-13:
I suggest you to have a look at gadget Bridge they have all the protocol implemented. Be aware that mi band starting from model 4 encrypts all the communication there is no way to get the key if you don't communicate with the servers of xiaomi.